Screenly/Anthias #535: disabling overscan stops automatic login

fakegold, 2017-01-06:
Running latest non-dev branch on pi2. 

When I disable overscan, screenly splash screen pops up, but then turns off, and instead goes through the lines in the image I've posted and stops at "raspberrypi login:_"

Turning overscan back on doesn't seem to fix the problem. 

[image]

vpetersson, 2017-01-06:
It seems unlikely that overscan would affect this.

How are you disabling the overscan?

fakegold, 2017-01-06:
It seems very unlikely, yet it is consistently the thing that I do that causes this to happen. I'm using raspi-config. I'm on my third installation and it keeps happening. Everything is fine until I disable overscan. 

When the pi reboots the screenly splash page finally fills the whole screen, then...the attached screenshot. 

It could very well be something else, but it happens when I disable overscan. Since I don't know how to fix it, it's back to the drawing board each time.

vpetersson, 2017-01-06:
> It seems very unlikely, yet it is consistently the thing that I do that causes this to happen.

Is this three different SD cards/devices or on the same device?

fakegold, 2017-01-06:
One SD card it has happened twice on (brand new out of the box card) and one on a card that i've been running with a screenly installation for the last two+ years without a problem. I installed a new version on both and that is when the problem arose.  

I just left overscan on on the older one and it's running fine. I need to reinstall the system on the other card again this morning to see where I'm at with that one.

vpetersson, 2017-01-06:
Try switching to the master/development branch to see if that solves the issue.

fakegold, 2017-01-06:
Okay, I have tested this with .9 and .9.1 now.

I cannot duplicate the problem with .9 (I'll use that one for now)

But can repeatedly produce the problem with .9.1

vpetersson, 2017-01-06:
If you install 0.9 and then run the upgrade, are you still able to reproduce it then? If not, it could be something with the disk image.

fakegold, 2017-01-06:
Upgrade is complete and pi is running with no overscan and looks great. I didn't reset the overscan, so I'm not sure if it's a disk image issue or not. Upon reboot it...still works! 

i loaded the .9.1 image from two different downloads onto the SD cards before, so not sure if that eliminates an issue with the image or not. Regardless, this troubleshoots around the issue.

fakegold, 2017-01-06:
Updating .9.1 and turning overscan off reproduces the problem on my system as well.

vpetersson, 2017-01-07:
Very interesting. I wonder if it is a bug in one of the software components. When you ran the upgrade, did you also answer Yes to the question of upgrading all the system? If so, let's try without that.

fakegold, 2017-01-07:
I answered No both times. Those systems are both running great right now. One upgraded to .9.1 and one at .9 that will I'll upgrade in the next day or two. I have another system running a build from two and a half years ago that I'm gonna jump on next. Hopefully, just upgrading it will go smoothly.

vpetersson, 2017-01-07:
Alright. If you answered no both times, let's try answering yes on one of them to see if that changes anything.

fakegold, 2017-01-10:
Updated the .9 system to .9.1 via ssh. Chose dev build and updated system. Everything seems to be running just fine. Overscan was already turned off before upgrade and continues to be working just fine.  I have a third machine I need to upgrade in the next week or so as well, will advise as I proceed with that too.

vpetersson, 2017-01-11:
Cool. So we know that it's something between 9.0 and 9.1. odd that we're unable to pin-point it, but it could be something in the image itself.

We'll release a new disk when [Sprint 2](https://github.com/wireload/screenly-ose/milestone/4) is done.

vpetersson, 2017-01-13:
Closing out since it has been resolved in master.

closed 2017-01-13 by vpetersson