Screenly/Anthias #536: RPi B+ gets blank screen after boot

DShearon, 2017-01-13:
I flashed the card with the latest image and booted the pi. I can see the initial boot up and splash screen and then I get a blank screen after that. I have confirmed that the pi has network connectivity as I am able to SSH in as well as access the web portal. 

I ran tvservice to confirm that the monitor I have it connected to supports the mode it is using. 

Anyone have any ideas?

DShearon, 2017-01-13:
Just wanted to update that after I sent this I went ahead and updated to the developer branch and it appears to be working fine.

vpetersson, 2017-01-13:
Thanks for reporting @DShearon. It's likely related to #535. Closing it out as it has been resolved in master.

closed 2017-01-13 by vpetersson