Screenly/Anthias #537: White Line appearing on Webpage Assets

madcackler, 2017-01-16:
With the current release of screenly, when I display a webpage there is a white line that appears at the bottom of the screen, no matter what the webpage source.  If I display an image this white line
[image]

 does not appear.

vpetersson, 2017-01-16:
My best guess is that this is an overscan issue. Try modifying `overscan_bottom` in `config.txt` to see if that solves the issue.

madcackler, 2017-01-18:
Currently I have overscan disabled.  The key thing is though, this only occurs when I display any webpages, and not just my own; if I display an image, it goes to edge no problem, so I suspect it has something to do with the built in browser's rendering.  My previous builds with screenly have not had this issue.

vpetersson, 2017-01-19:
@madcackler Are you able to reproduce this on *every* web page? Because images are using the browser as well.

madcackler, 2017-01-19:
Outside of pages that have a white background (in which it blends in), it appears visible on every page I've tried.

closed 2018-10-19 by stale[bot]