Screenly/Anthias #538: Unable to add HTTPS asset

DShearon, 2017-01-26:
We recently secured our Nagios server with SSL and I am now unable to add the page as an asset. It worked fine before we implemented SSL and if I enter the URL into any desktop browser it opens without issue. Any help or suggestions would be appreciated. The error I am getting is listed below.

Server Error: Could not retrieve file. Check the asset URL.

DShearon, 2017-01-26:
After some digging I found this was related to issue #234 

I can report that setting the verify_ssl=False did in fact fix the problem. 

Can any one recommend a way to quickly update the CA's without doing it manually?

vpetersson, 2017-01-27:
Are you running on Wheezy or Jessie?

DShearon, 2017-01-27:
I tried running the install from a clean Jessi-lite install but it would never auto-login and also failed the Screenly update during the DB migration portion. I am trying the built image at the moment from sprint 1. It accepted the URL but auto-login is failing. I am running the update and moving to the development branch to see if that helps.

DShearon, 2017-01-27:
Here is the failure message from DB migration during the last update.

TASK [screenly : Run database migration] ***************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:01.230790", "end": "2017-01-27 15:02:18.407015", "failed": true, "rc": 1, "start": "2017-01-27 15:02:17.176225", "stderr": "Traceback (most recent call last):\n  File \"/home/pi/screenly/bin/migrate.py\", line 158, in <module>\n    migrate_drop_filename()\n  File \"/home/pi/screenly/bin/migrate.py\", line 148, in migrate_drop_filename\n    with open_db_get_cursor() as (cursor, _):\n  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__\n    return self.gen.next()\n  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor\n    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:\nsqlite3.OperationalError: unable to open database file", "stdout": "", "stdout_lines": [], "warnings": []}

vpetersson, 2017-01-30:
I'd probably say it's likely a corrupted SD card. Try swapping it out. Unfortunately these SD cards are very  unreliable.

chriswier, 2017-02-10:
I'll chime in with a "me too" on the migrate.py issue.  New Raspbian Jessie Lite install (2017-01-11) version, fully updated system on a Raspi 3 model B.  I've tried two different SD cards at this time, and see the same issue for both.

john-arvid, 2017-02-18:
Same problem here. Clean install of Raspian Lite (through noob), get the same error. Tested with confirmed good SD cards.

vpetersson, 2017-04-13:
I'm unable to reproduce this. Tried a fresh Raspbian Lite (through Noob too) and was able to install things just fine.  Closing this one out until we can reproduce it.

closed 2017-04-13 by vpetersson

urb5tar, 2017-07-04:
I had the same issue today. I figured out, that root does not have sufficent rights to migrate the database. After i have switched to the user pi it worked.

vpetersson, 2017-07-04:
Hmm that's odd. Can you provide me with the output of:
```
ls -lah ~/.screenly/screenly.db
```

urb5tar, 2017-07-04:
-rw-r--r-- 1 pi pi 5.0K Jul  4 09:50 /home/pi/.screenly/screenly.db