Screenly/Anthias #540: Default install is not working (Big Buck Bunny)

Michael-Luthi, 2017-02-03:
Hi,

On raspberry Pi v1 & v2, the 2016-12-01 version is not working ! I haven't test with raspberry Pi v3.

When I turn on the Big buck bunny video, the player crashed with this error :

[image]

vpetersson, 2017-02-03:
Try upgrading to the latest development version to see if that solves the issue.

Michael-Luthi, 2017-02-03:
Sorry but I'm not used to GitHub so can you please explain me how to install the latest development version ?
Thank you

vpetersson, 2017-02-03:
Just SSH into the box and run:

```
bash <(curl -sL https://www.screenly.io/install-ose.sh)
```

Then answer 'yes' to the question if you want to install the development version.

Michael-Luthi, 2017-02-03:
After 4 hours installing the dev version, it seems to work !

Since the new release, I've also an bug after the splashcreen where screenly stays in command line and doesn't switch into the "viewer mode". It did not happen all the time.

Thank you

ruielias, 2017-02-15:
I also have the same problem defaut installation.
Can you tell me the steps you've taken?
Thank you.

boschie, 2017-06-15:
> Just SSH into the box and run:

bash <(curl -sL https://www.screenly.io/install-ose.sh)
Then answer 'yes' to the question if you want to install the development version.

> 
doesn't seem to work for me. 
When i activate that command nothing happens.

What should i do to get rid of issue #454?

Hendrie

vpetersson, 2017-06-15:
Hmm very strange. 

Could you please post the full output of the run?

boschie, 2017-06-15:
There is no output (see the screenshot).

[image]

vpetersson, 2017-06-15:
Can you add a "-v" to the curl command?

boschie, 2017-06-16:
$ bash <(curl -v -sL https://www.screenly.io/install-ose.sh) gives:

```
* Hostname was NOT found in DNS cache
*   Trying 104.27.188.73...
* Connected to www.screenly.io (104.27.188.73) port 443 (#0)
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* SSLv3, TLS handshake, Client hello (1):
} [data not shown]
* SSLv3, TLS handshake, Server hello (2):
{ [data not shown]
* SSLv3, TLS handshake, CERT (11):
{ [data not shown]
* SSLv3, TLS handshake, Server key exchange (12):
{ [data not shown]
* SSLv3, TLS handshake, Server finished (14):
{ [data not shown]
* SSLv3, TLS handshake, Client key exchange (16):
} [data not shown]
* SSLv3, TLS change cipher, Client hello (1):
} [data not shown]
* SSLv3, TLS handshake, Finished (20):
} [data not shown]
* SSLv3, TLS change cipher, Client hello (1):
{ [data not shown]
* SSLv3, TLS handshake, Finished (20):
{ [data not shown]
* SSL connection using TLSv1.2 / ECDHE-ECDSA-AES128-GCM-SHA256
* Server certificate:
*        subject: OU=Domain Control Validated; OU=PositiveSSL Multi-Domain; CN=sni35563.cloudflaressl.com
*        start date: 2017-06-14 00:00:00 GMT
*        expire date: 2017-12-21 23:59:59 GMT
*        subjectAltName: www.screenly.io matched
*        issuer: C=GB; ST=Greater Manchester; L=Salford; O=COMODO CA Limited; CN=COMODO ECC Domain Validation Secure Server CA 2
*        SSL certificate verify ok.
> GET /install-ose.sh HTTP/1.1
> User-Agent: curl/7.38.0
> Host: www.screenly.io
> Accept: */*
>
< HTTP/1.1 302 Moved Temporarily
< Date: Fri, 16 Jun 2017 09:21:54 GMT
< Content-Type: text/html
< Transfer-Encoding: chunked
< Connection: keep-alive
< Set-Cookie: __cfduid=dd2596626dfde34963054482e0a516c141497604913; expires=Sat, 16-Jun-18 09:21:53 GMT; path=/; domain=.screenly.io; HttpOnly
< Location: https://raw.github.com/wireload/screenly-ose/master/bin/install.sh
* Server cloudflare-nginx is not blacklisted
< Server: cloudflare-nginx
< CF-RAY: 36fcb3986d392c4e-AMS
<
* Ignoring the response-body
{ [data not shown]
* Connection #0 to host www.screenly.io left intact
* Issue another request to this URL: 'https://raw.github.com/wireload/screenly-ose/master/bin/install.sh'
* Hostname was NOT found in DNS cache
*   Trying 151.101.36.133...
* Connected to raw.github.com (151.101.36.133) port 443 (#1)
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* SSLv3, TLS handshake, Client hello (1):
} [data not shown]
* SSLv3, TLS handshake, Server hello (2):
{ [data not shown]
* SSLv3, TLS handshake, CERT (11):
{ [data not shown]
* SSLv3, TLS alert, Server hello (2):
} [data not shown]
* SSL certificate problem: unable to get local issuer certificate
* Closing connection 1
* SSLv3, TLS alert, Client hello (1):
} [data not shown]

```

vpetersson, 2017-06-16:
Hmm, it's could be an SSL issue if you're running an older version of Raspbian.

Try this:

```
sudo apt-get update
sudo apt-get upgrade
cd ~/screenly
git pull
```

boschie, 2017-06-16:
I'm using jessie Linux raspberrypi 4.9.24+

I'm doing the update now....

boschie, 2017-06-16:
Upgraded... git pull gave: Already up-to-date.

vpetersson, 2017-06-16:
ok now try:

```
cd ~/screenly/bin
./install.sh
```

boschie, 2017-06-16:
$ bash <(curl -v -sL https://www.screenly.io/install-ose.sh) still shows the same output...

boschie, 2017-06-16:
Every answer should be "y" ?

vpetersson, 2017-06-16:
Correct.

boschie, 2017-06-16:
It says:
`Installation completed`

A reboot would be the thing to do now?

vpetersson, 2017-06-16:
Can you post the full output of the run?

boschie, 2017-06-16:
How should i do this?

boschie, 2017-06-16:
Okay, just copied and paste here...

```
pi@raspberrypi:~/screenly/bin $ ./install.sh
   _____                           __         ____  _____ ______
  / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
  \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
 ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
/____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                 /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Would you like to use the development branch? You will get the latest features, but thi                       ngs may break. (y/N)

Would you like to perform a full system upgrade as well? (y/N)
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt-get update
Get:1 http://mirrordirector.raspbian.org jessie InRelease [14,9 kB]
Hit http://archive.raspberrypi.org jessie InRelease
Get:2 http://mirrordirector.raspbian.org jessie/main armhf Packages [9533 kB]
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-nl_NL
Ign http://archive.raspberrypi.org jessie/main Translation-nl
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-nl_NL
Ign http://archive.raspberrypi.org jessie/ui Translation-nl
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:3 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [43,3 kB]
Get:4 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [84,2 kB]
Get:5 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-nl_NL
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-nl
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-nl_NL
Ign http://mirrordirector.raspbian.org jessie/main Translation-nl
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-nl_NL
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-nl
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-nl_NL
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-nl
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9677 kB in 1min 46s (91,1 kB/s)
Reading package lists... Done
+ sudo apt-get purge -y python-setuptools python-pip python-pyasn1
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'python-pip' is not installed, so not removed
Package 'python-pyasn1' is not installed, so not removed
Package 'python-setuptools' is not installed, so not removed
The following package was automatically installed and is no longer required:
  libllvm3.7
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ sudo apt-get install -y python-dev git-core libffi-dev libssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
git-core is already the newest version.
libffi-dev is already the newest version.
libssl-dev is already the newest version.
python-dev is already the newest version.
[... 272 lines cut ...]
Del tzdata 2016j-0+deb8u1 [186 kB]
Del libudev1 215-17+deb8u6 [53,3 kB]
Del bind9-host 1:9.9.5.dfsg-9+deb8u10 [65,6 kB]
Del samba-common 2:4.2.14+dfsg-0+deb8u2 [269 kB]
Del libc-dev-bin 2.19-18+deb8u7 [231 kB]
Del locales 2.19-18+deb8u7 [3946 kB]
Del liblwres90 1:9.9.5.dfsg-9+deb8u10 [47,8 kB]
Del libsmbclient 2:4.2.14+dfsg-0+deb8u2 [138 kB]
Del samba-libs 2:4.2.14+dfsg-0+deb8u5 [4249 kB]
Del libisccfg90 1:9.9.5.dfsg-9+deb8u9 [50,4 kB]
Del libisccfg90 1:9.9.5.dfsg-9+deb8u10 [50,3 kB]
Del libgnutls-deb0-28 3.3.8-6+deb8u4 [625 kB]
Del libisc-export95 1:9.9.5.dfsg-9+deb8u9 [124 kB]
+ sudo apt-get clean
+ sudo find /usr/share/doc -depth -type f '!' -name copyright -delete
+ sudo find /usr/share/doc -empty -delete
+ sudo rm -rf /usr/share/man /usr/share/groff /usr/share/info /usr/share/lintian /usr/share/linda /var/cache/man
+ sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -exec rm -r '{}' ';'
+ cd /home/pi/screenly
+ git rev-parse HEAD
+ set +x
Installation completed.

```

vpetersson, 2017-06-16:
Great. Looks like it works. What do you get for `ps faux | grep python`.

boschie, 2017-06-16:
```
pi@raspberrypi:~/screenly/bin $ ps faux | grep python
pi       18847  2.0  0.4   4508  1992 pts/1    S+   14:50   0:00              \_ grep --color=auto python
root       721  0.2  1.2   9868  5528 ?        Ss   00:18   2:19 /usr/bin/python /home/pi/OnOffMotionDetection.py
pi       18276  0.5  4.4  24712 19960 ?        Ss   14:02   0:14 /usr/bin/python /home/pi/screenly/server.py
pi       18310  7.2  3.6  24712 16144 ?        S    14:03   3:22  \_ /usr/bin/python /home/pi/screenly/server.py
pi       18428 21.8  4.3  58456 19200 ?        Ssl  14:04   9:57 /usr/bin/python /home/pi/screenly/viewer.py
pi       18475  0.3  1.6  13132  7328 ?        S    14:05   0:10 python /usr/bin/uzbl-event-manager -va start

```

vpetersson, 2017-06-16:
Looks like everything is running fine then.

boschie, 2017-06-16:
Okay. But how should keep up to date?

vpetersson, 2017-06-16:
Just repeat the install script.

closed 2017-06-16 by vpetersson

boschie, 2017-06-16:
This last `./install.sh` ?

vpetersson, 2017-06-16:
Yes.

boschie, 2017-06-16:
Okay, thanks for the help. You are doing a great job!

vpetersson, 2017-06-16:
Thanks, but lots of people put in a lot of effort into this, so I don't want to take any credit from them. :)