Screenly/Anthias #541: Screen going back after some assets

gajjar8055, 2017-02-06:
eb 06 16:28:50 raspberrypi python[767]: Sleeping for 10
Feb 06 16:29:00 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:29:00 raspberrypi python[767]: get_next_asset
Feb 06 16:29:00 raspberrypi python[767]: refresh_playlist
Feb 06 16:29:00 raspberrypi python[767]: refresh: counter: (1) deadline (2017-02-10 10:35:00) timecur (2017-02-06 10:59:00.073180)
Feb 06 16:29:00 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:29:00 raspberrypi python[767]: get_next_asset counter 1 returning asset 1 of 2
Feb 06 16:29:00 raspberrypi python[767]: Showing asset Flashtr_V01_1.mp4 (video)
Feb 06 16:29:00 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750
Feb 06 16:29:00 raspberrypi python[767]: Displaying video /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750 for N/A
Feb 06 16:29:00 raspberrypi python[767]: current url is file:///tmp/screenly_html/black_page.html
Feb 06 16:29:43 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:29:43 raspberrypi python[767]: get_next_asset
Feb 06 16:29:43 raspberrypi python[767]: refresh_playlist
Feb 06 16:29:43 raspberrypi python[767]: refresh: counter: (1) deadline (2017-02-10 10:35:00) timecur (2017-02-06 10:59:43.213249)
Feb 06 16:29:43 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:29:43 raspberrypi python[767]: get_next_asset counter 1 returning asset 2 of 2
Feb 06 16:29:43 raspberrypi python[767]: Showing asset 14734165741473227513Alphathum-App-Img.jpg (image)
Feb 06 16:29:43 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/909d1d71057d4afc89ece5179ce3b315
Feb 06 16:29:43 raspberrypi python[767]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Feb 06 16:29:43 raspberrypi python[767]: Sleeping for 10
Feb 06 16:29:53 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:29:53 raspberrypi python[767]: get_next_asset
Feb 06 16:29:53 raspberrypi python[767]: refresh_playlist
Feb 06 16:29:53 raspberrypi python[767]: refresh: counter: (2) deadline (2017-02-10 10:35:00) timecur (2017-02-06 10:59:53.292764)
Feb 06 16:29:53 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:29:53 raspberrypi python[767]: get_next_asset counter 2 returning asset 1 of 2
Feb 06 16:29:53 raspberrypi python[767]: Showing asset Flashtr_V01_1.mp4 (video)
Feb 06 16:29:53 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750
Feb 06 16:29:53 raspberrypi python[767]: Displaying video /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750 for N/A
Feb 06 16:29:53 raspberrypi python[767]: current url is file:///tmp/screenly_html/black_page.html
Feb 06 16:30:36 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:30:36 raspberrypi python[767]: get_next_asset
Feb 06 16:30:36 raspberrypi python[767]: refresh_playlist
Feb 06 16:30:36 raspberrypi python[767]: refresh: counter: (2) deadline (2017-02-10 10:35:00) timecur (2017-02-06 11:00:36.447943)
Feb 06 16:30:36 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:30:36 raspberrypi python[767]: get_next_asset counter 2 returning asset 2 of 2
Feb 06 16:30:36 raspberrypi python[767]: Showing asset 14734165741473227513Alphathum-App-Img.jpg (image)
Feb 06 16:30:36 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/909d1d71057d4afc89ece5179ce3b315
Feb 06 16:30:36 raspberrypi python[767]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Feb 06 16:30:36 raspberrypi python[767]: Sleeping for 10
Feb 06 16:30:46 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:30:46 raspberrypi python[767]: get_next_asset
Feb 06 16:30:46 raspberrypi python[767]: refresh_playlist
Feb 06 16:30:46 raspberrypi python[767]: refresh: counter: (3) deadline (2017-02-10 10:35:00) timecur (2017-02-06 11:00:46.517032)
Feb 06 16:30:46 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:30:46 raspberrypi python[767]: get_next_asset counter 3 returning asset 1 of 2
Feb 06 16:30:46 raspberrypi python[767]: Showing asset Flashtr_V01_1.mp4 (video)
Feb 06 16:30:46 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750
Feb 06 16:30:46 raspberrypi python[767]: Displaying video /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750 for N/A
Feb 06 16:30:46 raspberrypi python[767]: current url is file:///tmp/screenly_html/black_page.html
Feb 06 16:31:29 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:31:29 raspberrypi python[767]: get_next_asset
Feb 06 16:31:29 raspberrypi python[767]: refresh_playlist
Feb 06 16:31:29 raspberrypi python[767]: refresh: counter: (3) deadline (2017-02-10 10:35:00) timecur (2017-02-06 11:01:29.721349)
Feb 06 16:31:29 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:31:29 raspberrypi python[767]: get_next_asset counter 3 returning asset 2 of 2
Feb 06 16:31:29 raspberrypi python[767]: Showing asset 14734165741473227513Alphathum-App-Img.jpg (image)
Feb 06 16:31:29 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/909d1d71057d4afc89ece5179ce3b315
Feb 06 16:31:29 raspberrypi python[767]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Feb 06 16:31:29 raspberrypi python[767]: Sleeping for 10
Feb 06 16:31:39 raspberrypi python[767]: Last update: 2017-02-06 16:02:18.253803
Feb 06 16:31:39 raspberrypi python[767]: get_next_asset
Feb 06 16:31:39 raspberrypi python[767]: refresh_playlist
Feb 06 16:31:39 raspberrypi python[767]: refresh: counter: (4) deadline (2017-02-10 10:35:00) timecur (2017-02-06 11:01:39.777440)
Feb 06 16:31:39 raspberrypi python[767]: get_next_asset after refresh
Feb 06 16:31:39 raspberrypi python[767]: get_next_asset counter 4 returning asset 1 of 2
Feb 06 16:31:39 raspberrypi python[767]: Showing asset Flashtr_V01_1.mp4 (video)
Feb 06 16:31:39 raspberrypi python[767]: Asset URI /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750
Feb 06 16:31:39 raspberrypi python[767]: Displaying video /home/pi/screenly_assets/6050a3794ba54d958ca4af205a6df750 for N/A
Feb 06 16:31:39 raspberrypi python[767]: current url is file:///tmp/screenly_html/black_page.html

Feb 06 16:32:41 raspberrypi sshd[1129]: Accepted password for pi from 192.168.1.103 port 63496 ssh2
Feb 06 16:32:41 raspberrypi sshd[1129]: pam_unix(sshd:session): session opened for user pi by (uid=0)
Feb 06 16:32:41 raspberrypi systemd[1]: Starting Session c2 of user pi.
Feb 06 16:32:41 raspberrypi systemd-logind[453]: New session c2 of user pi.
Feb 06 16:32:41 raspberrypi systemd[1]: Started Session c2 of user pi.

KarlReimer, 2017-02-06:
I believe I'm having a very similar problem. My Pi/Screenly will work fine with a basic set of .mp4 assets indefinitely. Then, if I add in some particular .JPG files, Screenly will work for a while, successfully loading and displaying those JPGs, but then at some later point hang when loading one of these newly added JPGs and just show the blank_page.html file. Attaching a screenshot of the web interface logs.

[image]

Could you tell me how you get the logs above? I would like to get the logs off of my Pi to see if they are similar to the above.

gajjar8055, 2017-02-07:
@KarlReimer `sudo journalctl -f` this command for log

vpetersson, 2017-02-08:
@gajjar8055 

Can you provide the output of the following commands?

```
$ ps faux
$ free -m
$ uptime
```

I'm thinking that it might be a load or memory issue.

gajjar8055, 2017-02-11:
[dump.txt](https://github.com/wireload/screenly-ose/files/768229/dump.txt)
here i have attached all require command output please check.

gajjar8055, 2017-02-11:
and the new issue arrived today morning. 
[image]
active asset not gonna place  in active asset tab.

vpetersson, 2017-02-12:
@gajjar8055 That's because the active assets have an expiration date in the past.

gajjar8055, 2017-02-16:
@vpetersson what about black screen appear after some assets. is it my problem or its occurring somewhere else also?

vpetersson, 2017-04-13:
@gajjar8055 

> what about black screen appear after some assets. is it my problem or its occurring somewhere else also?

Assuming this is for videos, it's expected. It can take a second or two to load a video.

vpetersson, 2017-04-13:
@KarlReimer Are you able to reproduce this if you upgrade to the latest development version?

KarlReimer, 2017-04-15:
Actually, I am not sure I ever had this bug. The problem I was having is better described in ticket #542, which I opened and have since closed. I have not had any problems with our screenly installs since then. I also have not upgraded Screenly.

vpetersson, 2017-04-15:
@KarlReimer Got it. Closing out this issue then.

closed 2017-04-15 by vpetersson