Screenly/Anthias #542: Screenly goes black, stops displaying assets

KarlReimer, 2017-02-07:
After a seemingly random amount of time, (varying from minutes to hours, in one case as long as 36 hours,) Screenly will go black and stop displaying assets. Looking at the logs, it shows that "current url is file:///tmp/screenly_html/black_page.html" but it does not seem to load the next asset. All assets have been successfully shown in sequence and repeated at least once when this happens.

I commented on issue #541, it's not clear to me if that is the same issue as this.

This same Pi/Screenly install was working fine previously when testing with a computer monitor and some basic images, though I suppose because of the variable nature of how long it takes for the assets to stop loading I can't be completely sure that this same issue would not have happened in that environment. Since then we have rolled it out to a different building using a Samsung model #UN40J5200AFXZC, and added some new assets.

Additional note; 10.1.1.17 is a monitoring service.

```
Feb 07 14:43:31 telmRasPi01 python[730]: Showing asset Thrift Banner video 1 FINAL.mp4 (video)
Feb 07 14:43:31 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:43:37 telmRasPi01 sshd[3021]: Connection closed by 10.1.1.17 [preauth]
Feb 07 14:44:03 telmRasPi01 python[730]: Showing asset Captioned version-About MCC (Canada version).mov (video)
Feb 07 14:44:03 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:45:56 telmRasPi01 python[730]: Showing asset Thrift Banner Video 2 Final.mp4 (video)
Feb 07 14:45:56 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:46:28 telmRasPi01 python[730]: Showing asset NEWMcc Food Programs (1).mp4 (video)
Feb 07 14:46:28 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:47:55 telmRasPi01 python[730]: Showing asset Thrift Banner Video 3 Final.mp4 (video)
Feb 07 14:47:55 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:48:13 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 14:48:25 telmRasPi01 python[730]: Showing asset NEWmcceducationprograms.mp4 (video)
Feb 07 14:48:25 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:48:37 telmRasPi01 sshd[3166]: Connection closed by 10.1.1.17 [preauth]
Feb 07 14:49:59 telmRasPi01 python[730]: Showing asset Thrift Banner Video 4 Final.mp4 (video)
Feb 07 14:50:00 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:50:32 telmRasPi01 python[730]: Showing asset NEWVideo-partners2017.mp4 (video)
Feb 07 14:50:32 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:51:46 telmRasPi01 python[730]: Showing asset Thrift Banner video 1 FINAL.mp4 (video)
Feb 07 14:51:46 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:51:59 telmRasPi01 wpa_supplicant[377]: wlan0: WPA: Group rekeying completed with 32:b5:c2:c8:83:e0 [GTK=CCMP]
Feb 07 14:52:18 telmRasPi01 python[730]: Showing asset Captioned version-About MCC (Canada version).mov (video)
Feb 07 14:52:18 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:53:38 telmRasPi01 sshd[3273]: Connection closed by 10.1.1.17 [preauth]
Feb 07 14:53:54 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 14:54:11 telmRasPi01 python[730]: Showing asset Thrift Banner Video 2 Final.mp4 (video)
Feb 07 14:54:11 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:54:43 telmRasPi01 python[730]: Showing asset NEWMcc Food Programs (1).mp4 (video)
Feb 07 14:54:43 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:56:10 telmRasPi01 python[730]: Showing asset Thrift Banner Video 3 Final.mp4 (video)
Feb 07 14:56:10 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:56:40 telmRasPi01 python[730]: Showing asset NEWmcceducationprograms.mp4 (video)
Feb 07 14:56:41 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:58:15 telmRasPi01 python[730]: Showing asset Thrift Banner Video 4 Final.mp4 (video)
Feb 07 14:58:15 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 14:58:39 telmRasPi01 sshd[3420]: Connection closed by 10.1.1.17 [preauth]
Feb 07 14:58:47 telmRasPi01 python[730]: Showing asset NEWVideo-partners2017.mp4 (video)
Feb 07 14:58:47 telmRasPi01 python[730]: current url is file:///tmp/screenly_html/black_page.html
Feb 07 15:01:59 telmRasPi01 wpa_supplicant[377]: wlan0: WPA: Group rekeying completed with 32:b5:c2:c8:83:e0 [GTK=CCMP]
Feb 07 15:03:28 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 15:03:40 telmRasPi01 sshd[3464]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:06:51 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 15:08:41 telmRasPi01 sshd[3486]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:11:59 telmRasPi01 wpa_supplicant[377]: wlan0: WPA: Group rekeying completed with 32:b5:c2:c8:83:e0 [GTK=CCMP]
Feb 07 15:12:24 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 15:13:41 telmRasPi01 sshd[3505]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:17:01 telmRasPi01 CRON[3511]: pam_unix(cron:session): session opened for user root by (uid=0)
Feb 07 15:17:01 telmRasPi01 CRON[3515]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb 07 15:17:01 telmRasPi01 CRON[3511]: pam_unix(cron:session): session closed for user root
Feb 07 15:18:42 telmRasPi01 sshd[3527]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:19:16 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
Feb 07 15:21:59 telmRasPi01 wpa_supplicant[377]: wlan0: WPA: Group rekeying completed with 32:b5:c2:c8:83:e0 [GTK=CCMP]
Feb 07 15:23:42 telmRasPi01 sshd[3545]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:28:43 telmRasPi01 sshd[3550]: Connection closed by 10.1.1.17 [preauth]
Feb 07 15:28:54 telmRasPi01 avahi-daemon[541]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fdd3:ba5e:c36e:0:ba27:ebff:feb1:68ad.
Feb 07 15:28:54 telmRasPi01 avahi-daemon[541]: Joining mDNS multicast group on interface wlan0.IPv6 with address fdd3:ba5e:c36e:0:8d74:c9b7:acad:f9c7.
Feb 07 15:28:55 telmRasPi01 ntpd[699]: Deleting interface #6 wlan0, fdd3:ba5e:c36e:0:ba27:ebff:feb1:68ad#123, interface stats: received=0, sent=0, dropped=0, active_time=7187 secs
Feb 07 15:28:55 telmRasPi01 ntpd[699]: peers refreshed
Feb 07 15:29:01 telmRasPi01 dhcpcd[674]: wlan0: fe80::32b5:c2ff:fec8:83e0 router available
```

KarlReimer, 2017-02-07:
Also could someone tell me how to turn debug logging on? Thanks!

vpetersson, 2017-02-08:
@KarlReimer If you're able to reproduce this, can you run the following commands *after* the screen goes black and post the output:

 ```
$ ps faux
$ free -m
$ uptime
```

KarlReimer, 2017-02-08:
Sorry I closed this accidentally when replying.

Below is the log from this morning's hang, and then the requested commands afterwards.

A bit more extra info:
First, I changed the background colour on the black_page.html to be white. Now when the Pi stops displaying assets we get a white screen. I did this just to confirm that it was indeed the Pi hanging and some other loss of video.

Second, I have been experimenting with different asset lists. I haven't done extensive testing (the hanging happens at such varying times it's hard to be sure that a list is "ok") but I think I may have identified a problem asset:

Captioned version-About MCC (Canada version).mov (video)

In the logs above, you can see that this video is played successfully a few times. However, when I ran a playlist without it, it succeeded in running fine for 24 hours. This morning, as SOON as I added that video back in, it hung again, which is what the logs below are from. You can see I added the asset, and then after that the playlist stopped again.

`Feb 08 14:52:12 telmRasPi01 python[816]: get_next_asset counter 0 returning asset 1 of 7
Feb 08 14:52:12 telmRasPi01 python[816]: Showing asset Thrift Banner video 1 FINAL.mp4 (video)
Feb 08 14:52:12 telmRasPi01 python[816]: Asset URI /home/pi/screenly_assets/d98d6c6968d743c9803ddf95189f2943
Feb 08 14:52:12 telmRasPi01 python[816]: Displaying video /home/pi/screenly_assets/d98d6c6968d743c9803ddf95189f2943 for N/A
Feb 08 14:52:12 telmRasPi01 python[816]: current url is file:///tmp/screenly_html/black_page.html
Feb 08 14:52:44 telmRasPi01 python[816]: Last update: 2017-02-08 13:30:12.227638
Feb 08 14:52:44 telmRasPi01 python[816]: get_next_asset
Feb 08 14:52:44 telmRasPi01 python[816]: refresh_playlist
Feb 08 14:52:44 telmRasPi01 python[816]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-08 14:52:44.932861)
Feb 08 14:52:44 telmRasPi01 python[816]: get_next_asset after refresh
Feb 08 14:52:44 telmRasPi01 python[816]: get_next_asset counter 0 returning asset 2 of 7
Feb 08 14:52:44 telmRasPi01 python[816]: Showing asset Thrift Banner Video 2 Final.mp4 (video)
Feb 08 14:52:44 telmRasPi01 python[816]: Asset URI /home/pi/screenly_assets/50bbb6f7a8b9487fbce65b2a270ba05e
Feb 08 14:52:44 telmRasPi01 python[816]: Displaying video /home/pi/screenly_assets/50bbb6f7a8b9487fbce65b2a270ba05e for N/A
Feb 08 14:52:44 telmRasPi01 python[816]: current url is file:///tmp/screenly_html/black_page.html
Feb 08 14:53:17 telmRasPi01 python[816]: Last update: 2017-02-08 13:30:12.227638
Feb 08 14:53:17 telmRasPi01 python[816]: get_next_asset
Feb 08 14:53:17 telmRasPi01 python[816]: refresh_playlist
Feb 08 14:53:17 telmRasPi01 python[816]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-08 14:53:17.063389)
Feb 08 14:53:17 telmRasPi01 python[816]: get_next_asset after refresh
Feb 08 14:53:17 telmRasPi01 python[816]: get_next_asset counter 0 returning asset 3 of 7
Feb 08 14:53:17 telmRasPi01 python[816]: Showing asset Thrift Banner Video 3 Final.mp4 (video)
Feb 08 14:53:17 telmRasPi01 python[816]: Asset URI /home/pi/screenly_assets/09c39d5b0bcd4e93ac34b0d5a3a48476
Feb 08 14:53:17 telmRasPi01 python[816]: Displaying video /home/pi/screenly_assets/09c39d5b0bcd4e93ac34b0d5a3a48476 for N/A
Feb 08 14:53:17 telmRasPi01 python[816]: current url is file:///tmp/screenly_html/black_page.html
Feb 08 14:53:47 telmRasPi01 python[816]: Last update: 2017-02-08 13:30:12.227638
Feb 08 14:53:47 telmRasPi01 python[816]: get_next_asset
Feb 08 14:53:47 telmRasPi01 python[816]: refresh_playlist
Feb 08 14:53:47 telmRasPi01 python[816]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-08 14:53:47.194744)
Feb 08 14:53:47 telmRasPi01 python[816]: get_next_asset after refresh
Feb 08 14:53:47 telmRasPi01 python[816]: get_next_asset counter 0 returning asset 4 of 7
Feb 08 14:53:47 telmRasPi01 python[816]: Showing asset Thrift Banner Video 4 Final.mp4 (video)
Feb 08 14:53:47 telmRasPi01 python[816]: Asset URI /home/pi/screenly_assets/19ab16fa303f46c9ab85cbb02c278877
Feb 08 14:53:47 telmRasPi01 python[816]: Displaying video /home/pi/screenly_assets/19ab16fa303f46c9ab85cbb02c278877 for N/A
Feb 08 14:53:47 telmRasPi01 python[816]: current url is file:///tmp/screenly_html/black_page.html
Feb 08 14:54:19 telmRasPi01 python[816]: Last update: 2017-02-08 13:30:12.227638
Feb 08 14:54:19 telmRasPi01 python[816]: get_next_asset
Feb 08 14:54:19 telmRasPi01 python[816]: refresh_playlist
Feb 08 14:54:19 telmRasPi01 python[816]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-08 14:54:19.317789)
Feb 08 14:54:19 telmRasPi01 python[816]: get_next_asset after refresh
Feb 08 14:54:19 telmRasPi01 python[816]: get_next_asset counter 0 returning asset 5 of 7
Feb 08 14:54:19 telmRasPi01 python[816]: Showing asset NEWMcc Food Programs (1).mp4 (video)
Feb 08 14:54:19 telmRasPi01 python[816]: Asset URI /home/pi/screenly_assets/f7b1839660bd46158e9d8b885ebba3ab
Feb 08 14:54:19 telmRasPi01 python[816]: Displaying video /home/pi/screenly_assets/f7b1839660bd46158e9d8b885ebba3ab for N/A
Feb 08 14:54:19 telmRasPi01 python[816]: current url is file:///tmp/screenly_html/black_page.html
Feb 08 14:54:59 telmRasPi01 sudo[19955]: pi : TTY=pts/2 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/python
Feb 08 14:55:46 telmRasPi01 python[816]: Last update: 2017-02-08 13:30:12.227638
Feb 08 14:55:46 telmRasPi01 python[816]: get_next_asset
Feb 08 14:55:46 telmRasPi01 python[816]: refresh_playlist
[... cut ...]
pi       19842  0.0  0.3   4976  3296 ?        Ss   14:53   0:00 /lib/systemd/systemd --user
pi       19847  0.0  0.1  24168  1332 ?        S    14:53   0:00  \_ (sd-pam)
pi@telmRasPi01:~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           925        427        497         18         45        242
-/+ buffers/cache:        139        786
Swap:            0          0          0
pi@telmRasPi01:~ $ uptime
 15:11:05 up 17:10,  1 user,  load average: 0.06, 0.13, 0.10`

vpetersson, 2017-02-08:
Hmm  nothing stick out at first glance. Wonder if it can be a faulty version of `omxplayer`. Could you try `sudo apt-get update && sudo apt-get -y upgrade`?

KarlReimer, 2017-02-10:
Okay so I ran the upgrades. Screenly continues to fail, but at this point 3/3 times it has failed, it has failed on that same .mov file.

Attached are the logs from this mornings failure, as well as the results from running the above commands (free - m etc)

[commandlogs_2017-02-10-9amest.txt](https://github.com/wireload/screenly-ose/files/766958/commandlogs_2017-02-10-9amest.txt)
[screenlylog_2017-02-10.txt](https://github.com/wireload/screenly-ose/files/766959/screenlylog_2017-02-10.txt)

I have removed the mov file from the asset list, and will leave Screenly running. I expect it will now run indefinitely, but will report back with the findings.

I find it interesting that the MOV file is able to be played several times, but then fails to load at some later point.

KarlReimer, 2017-02-10:
Okay no such luck - so much for my theory that a single asset was to blame.

Took out the asset, screenly froze again.

Feb 10 15:51:25 telmRasPi01 python[12110]: Last update: 2017-02-10 14:01:03.582131
Feb 10 15:51:25 telmRasPi01 python[12110]: get_next_asset
Feb 10 15:51:25 telmRasPi01 python[12110]: refresh_playlist
Feb 10 15:51:25 telmRasPi01 python[12110]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-10 15:51:25.282991)
Feb 10 15:51:25 telmRasPi01 python[12110]: get_next_asset after refresh
Feb 10 15:51:25 telmRasPi01 python[12110]: get_next_asset counter 0 returning asset 5 of 7
Feb 10 15:51:25 telmRasPi01 python[12110]: Showing asset NEWMcc Food Programs (1).mp4 (video)
Feb 10 15:51:25 telmRasPi01 python[12110]: Asset URI /home/pi/screenly_assets/f7b1839660bd46158e9d8b885ebba3ab
Feb 10 15:51:25 telmRasPi01 python[12110]: Displaying video /home/pi/screenly_assets/f7b1839660bd46158e9d8b885ebba3ab for N/A
Feb 10 15:51:25 telmRasPi01 python[12110]: current url is file:///tmp/screenly_html/black_page.html
Feb 10 15:52:52 telmRasPi01 python[12110]: Last update: 2017-02-10 14:01:03.582131
Feb 10 15:52:52 telmRasPi01 python[12110]: get_next_asset
Feb 10 15:52:52 telmRasPi01 python[12110]: refresh_playlist
Feb 10 15:52:52 telmRasPi01 python[12110]: refresh: counter: (0) deadline (2017-03-01 04:55:00) timecur (2017-02-10 15:52:52.474831)
Feb 10 15:52:52 telmRasPi01 python[12110]: get_next_asset after refresh
Feb 10 15:52:52 telmRasPi01 python[12110]: get_next_asset counter 0 returning asset 6 of 7
Feb 10 15:52:52 telmRasPi01 python[12110]: Showing asset NEWmcceducationprograms.mp4 (video)
Feb 10 15:52:52 telmRasPi01 python[12110]: Asset URI /home/pi/screenly_assets/1a3ff603a08f4c60ba0bff62f2766152
Feb 10 15:52:52 telmRasPi01 python[12110]: Displaying video /home/pi/screenly_assets/1a3ff603a08f4c60ba0bff62f2766152 for N/A
Feb 10 15:52:52 telmRasPi01 python[12110]: current url is file:///tmp/screenly_html/black_page.html

[commandlogs_2017-02-10-1230pmest.txt](https://github.com/wireload/screenly-ose/files/767413/commandlogs_2017-02-10-1230pmest.txt)

At this point, the only pattern I can see is that all the failures have been at the end of the playlist. Specifically the get next counter line is always showing one asset before the end, and then as it attempts to load the final asset it fails.

"Feb 10 15:52:52 telmRasPi01 python[12110]: get_next_asset counter 0 returning asset 6 of 7"

Maybe I'm just desperate to find a pattern here... I don't know.

KarlReimer, 2017-02-15:
I have gone back to a basic set of image and video assets that had been working previously. The Pi has now been successfully running for more than 48 hours.

NoWHereRepo, 2017-03-13:
We had the same issue. The hanging behaviour appeared only with some videos.

In our case the GPU memory config was not sufficient. We increased the GPU memory to 128MB in raspi-config. Since this change we have stable screenly raspis

vpetersson, 2017-03-13:
Go catch, @NoWHereRepo.

bcallies, 2017-03-24:
We had this same issue. Changing GPU memory to 128MB, per @NoWHereRepo 's note, resolved the issue (stable/no freezes for more than 2 days.

vpetersson, 2017-03-24:
We should add a check to the installer that ensures GPU mem is 128 or larger.

NoWHereRepo, 2017-03-24:
Still no freezes since 2 weeks

KarlReimer, 2017-03-27:
I believe that this is resolved for me as well thanks to @NoWHereRepo 's suggestion of changing the GPU memory allocation. All of our pi's were set to 64MB which I changed to 128MB.

closed 2017-03-27 by KarlReimer

cworobetz, 2017-07-25:
This issue was also affecting me (Using the NOOBs netinstall version of Screenly with latest apt upgrades), until I spent time digging on Google and found the fix suggested by @NoWHereRepo . I suggest that the GPU memory be checked / altered by Screenly as per @vpetersson 's suggestion so that others don't have the same experience I did trying to get this fixed.

vpetersson, 2017-08-03:
Good call, @cworobetz. Created ticket #606 to solve this.