Screenly/Anthias #543: Screenly’s management URL keep showing up

MrSinaRJ, 2017-02-13:
I have 3 assets which all are on and showing well, but after a few loops, the start page shows up with URL showing.
How to stop this from happening, this problem get more annoying when I add Big Buck Bunny movie to showing assets.

vpetersson, 2017-02-13:
There seem to be a few similar reports. I wonder if this is a bug related to OMXPlayer, which is what we are using for video playback.

MrSinaRJ, 2017-02-13:
Is there anything I can do to fix the problem?
I really loved VLC player for use in ubuntu, can I use that instead or it's a lot of hard work to do?

MrSinaRJ, 2017-02-14:
Something happened today and I didn't know if I should open a new issue or update this one related to video.

I tried (or I think I tried) to play 20 first seconds of the movie I uploaded to rpi, but it froze at first and then when I reload the management screen I saw the time I've entered there is disappeared.
Is this something related to this issue or it can't be done for movies?

antonmolodykh, 2017-04-07:
@MrSinaRJ Could you please provide exact steps to reproduce the issue?

vpetersson, 2017-04-11:
Closing this one out. Will re-open when we have steps to reproduce.

closed 2017-04-11 by vpetersson