Screenly/Anthias #544: Screenly-ose Does not work

ruielias, 2017-02-15:
I have a raspberry, I installed on the SD-Card Linux Jessie Lite. I did "apt-get update" then "apt-get upgrade" then the Screenly install command "$ bash <(curl -sL https://www.screenly.io/install-ose.sh)
".
I did reboot and I accessed raspberry ip: http: // ip:8080 and nothing appears.
What can be missing?

MrSinaRJ, 2017-02-18:
Hi
Did the "Screenly’s management page" appeared after rpi booted up? the page that shows your rpi IP to connect
there was no error during installation?

madhujiot, 2017-03-26:
hi, I've RPI3 B with wifi connected. I have Jessie lite installed and trying to execute the below and nothing happens. Any other step missing, please help, thanks.

root@raspberrypi:/home/pi# bash <(curl -sL --proto '=https' https://www.screenly.io/install-ose.sh)
root@raspberrypi:/home/pi# bash <(curl -sL --proto '=https' https://www.screenly.io/install-ose.sh)
root@raspberrypi:/home/pi#

vpetersson, 2017-03-26:
@madhujiot This is unrelated to this issue, but the issue is most likely related to that you're running the installer as root. It should be run as the user `pi`.

madhujiot, 2017-03-26:
ok vpetrsson, I can open another thread for the issue. Same result if I run as user pi.

vpetersson, 2017-04-05:
@madhujiot I was able to reproduce the issue you had when I had a typo in the URL. Could you post the output of:

```
curl -sL https://www.screenly.io/install-ose.sh
```

vpetersson, 2017-04-11:
Closing out due to lack of response. Will re-open if we get more information and/or can reproduce.

closed 2017-04-11 by vpetersson

darshinvyas, 2017-06-23:
hi VP,

I am trying to run Screenly as a different user, is it possible? 

Basically, the raspberry pi is on a windows domain. I have managed to get it connected to the windows AD and want to run Screenly as a Windows AD user (can bind it to one specific AD user). The reason is we do use SSO on a lot of web applications. I am wanting to open a browser link in Screenly, which at the moment wont open since it requires authentication. However, if I can run screenly as an AD user on my pi, it wont need browser authentication.

Basically, from your perspective, I need help with running Screenly as a different user. Is it possible and what config file should be changed? 

Thanks in advance.

Cheers,
Darshin

vpetersson, 2017-06-23:
@darshinvyas,

This is unrelated to the ticket you commented on. In short, it might be possible on a system level (with a fair amount of work), but that still won't solve your issue with the browser (as that's a separate issue).

darshinvyas, 2017-06-23:
yeah. i realised that. I have setup kerberos properly but still stuck with the browser issue. Wonder if you are planning to support in-browser auth?

vpetersson, 2017-06-23:
If by in-browser auth you mean like an arbitrary input field, then no. There is no way to do this well. Even companies like 1Password and Lastpass struggle with that.