Screenly/Anthias #545: Docker build error

azcoigreach, 2017-02-22:
I have tried to run and pull the Docker image from Docker automatically.  And I have tried building a Container from the Dockerfile manually.

Both ways seem to render the same error.

standard_init_linux.go:178: exec user process caused "exec format error"

Any help would be appreciated.  

Mahalo.

vpetersson, 2017-02-22:
I can't reproduce this. The image is built using Docker Hub, but builds just fine on my machine as well. 

```
~ ❯ docker run --rm -ti \                                                                                                                                                                                                                                                                                                                                                 ⏎
  -p 8080:8080 \
  wireload/screenly-ose-server
Bottle v0.12.9 server starting up (using GunicornServer(threads=2, timeout=20))...
Listening on http://0.0.0.0:8080/
Hit Ctrl-C to quit.

[2017-02-21 19:03:33 +0000] [5] [INFO] Starting gunicorn 19.4.5
[2017-02-21 19:03:33 +0000] [5] [INFO] Listening at: http://0.0.0.0:8080 (5)
[2017-02-21 19:03:33 +0000] [5] [INFO] Using worker: threads
[2017-02-21 19:03:33 +0000] [12] [INFO] Booting worker with pid: 12
```

What OS and Docker version are you using?

azcoigreach, 2017-02-22:
I tried building with a PI3 and Docker 1.13.1

After a little research last night I found a couple notes that debian:jessie is x86 and not ARMhf.  

Did you run this on a PI !?

---from your Dockerfile---
FROM debian:jessie  <<THIS DOESN'T WORK ON ANY PI. 
---

---result from the same command run on several PI's---
pi@sandbox:~ $ docker run --rm -ti -p 8080:8080 wireload/screenly-ose-server
Unable to find image 'wireload/screenly-ose-server:latest' locally
latest: Pulling from wireload/screenly-ose-server
75a822cd7888: Pull complete
c140fc03b940: Pull complete
c364de4a4453: Pull complete
76ab75d13f70: Pull complete
1f74dbdef0b3: Pull complete
c217ae7a7634: Pull complete
2221f18308c0: Pull complete
bf7d888965f6: Pull complete
b4070107b3db: Pull complete
Digest: sha256:4aadccda381e6181a449c4a0684273822c9ba35850382b6aff2ddce7f3712995
Status: Downloaded newer image for wireload/screenly-ose-server:latest
standard_init_linux.go:178: exec user process caused "exec format error"
---
--- result of running debian:jessie---
docker run -ti --rm debian:jessie
Unable to find image 'debian:jessie' locally
jessie: Pulling from library/debian
5040bd298390: Pull complete
Digest: sha256:abbe80c8c87b7e1f652fe5e99ff1799cdf9e0878c7009035afe1bccac129cad8
Status: Downloaded newer image for debian:jessie
standard_init_linux.go:178: exec user process caused "exec format error"
---

vpetersson, 2017-02-22:
No, you cannot run this on the Pi. That's not what it's designed for. It's designed as a handy tool for local development.

azcoigreach, 2017-02-22:
Well that's silly.  Docker this thing!!! Your figgin killing me here...

vpetersson, 2017-02-22:
We're all for Docker and are using it in Screenly Pro server-side. In this case, however, it would create too much overhead on the Raspberry Pi.

closed 2017-02-22 by vpetersson