Screenly/Anthias #546: Screenly OSE Active Asset Not Displaying

rono711, 2017-02-23:
Dear All,
I am 100% new in Screenly. I added asset and turned it ON. It's under Active Status. However, nothing is on my big screen? Any advice or tip would be much appreciated.

zaphod2010, 2017-02-23:
I had the same issue a while back., Turned out that the Raspberry's clock was way off.
I added this as a root cronjob:

`@reboot /bin/sleep 30 ; /bin/date -s "$(/usr/bin/curl -s --head http://google.com | /bin/grep ^Date: | /bin/sed 's/Date: //g')"`

vpetersson, 2017-04-05:
Just to chime in here. Normally, the `ntp` service should be responsible for setting the clock. As such, the root cause is likely either that NTP service is incorrectly configured, or NTP is blocked by the firewall. Since the Raspberry Pi doesn't have any RTC, NTP is required.

If you are unable to use NTP, I suggest you look into [HTP](http://www.vervest.org/htp/). I will close this issue as this is not related to Screenly per se, but rather a network issue.

closed 2017-04-05 by vpetersson