Screenly/Anthias #551: Scheduling doesn't work in latest OSE

jassoo, 2017-03-15:
Scheduling doesn't work in latest OSE (2016-12-01-Screenly_OSE_4GB).
It only works if i leave webbrowser open for OSE Screenly.
ex. if i have picture showing only today (ex. 01:00-23:59) and I want to schedule a new picture for tomorrow (ex. 08:00-23:59) It will show tomorrow 09:00 still old picture. If i open screenly OSE with webbrowser it "updates" everything as it should be.

jassoo, 2017-03-16:
I now tested older version (2016-08-11-Screenly_OSE_4GB.img) and same problem exists there.
Webbrowser must be open, otherwise scheduling assets doesn't work.

vpetersson, 2017-03-29:
I asked @antonmolodykh to look at this but was unable to reproduce it. 

@jassoo Could you please provide exact steps to reproduce the issue?

jassoo, 2017-03-29:
OK.
My setup: 
-PC with WIN10
-Raspberry Pi 3 model B
-2016-12-01-Screenly_OSE_4GB.img **OR** 2016-08-11-Screenly_OSE_4GB.img from github
-Win32Diskimager to "Burn" image

1. I "burn" image with Win32Diskimager to SD.
2. SD --> Raspberry Pi 3 model B
3. I open Screenly webpage with given DHCP ip.
4. I delete "demo" assets (pictures and webpage link)
5. I set ex. 2 new picture assets. ex.
            -PIC1 starting from this day from now-->stop tomorrow 08:00
            -PIC2 starting from tomorrow 09:00--> stop 21:00
6. I'll **CLOSE** webbrowser and check TV and its showing PIC1 as it should be.
7. I'll Check TV (**WITHOUT OPENING**Screenly in webbrowser first) tomorrow 09:30 and it still shows PIC1, but it should only show PIC2.
8. If i then open Screenly website it updates everything as it should be.

vpetersson, 2017-03-29:
Is the Raspberry Pi connected to the internet permanently? Also, could you check the output of `date` on the Pi?

(Pro-tip: switch to [Etcher](https://etcher.io/) for flashing out the disk image, as it does validation as well)

jassoo, 2017-03-29:
Yes, its connected to internet permanently and i have checked date and without changing it shows 2 hours off from where im living, but even changing timezone to right one, it doesn't make any difference to scheduling (date showing right time). I also thought this one first being the fail. 😄

vpetersson, 2017-03-29:
@antonmolodykh see if you can reproduce it using those steps.

antonmolodykh, 2017-03-31:
@jassoo we just fixed the problem. pleace, check if you can reproduce the bug after updating (not downloading the image, but according to option 2 in https://www.screenly.io/ose/)

jassoo, 2017-04-03:
Yes, scheduling is working if I install screenly using option 2.

vpetersson, 2017-04-03:
@jassoo Perfect. Then it will be fixed in the next disk image. Closing out this ticket.

closed 2017-04-03 by vpetersson