Screenly/Anthias #552: GTK-Warning cannot open display

dantheman2865, 2017-03-17:
Hello, I installed OSE from the SD image and it has stopped working after a week or so. When I reboot, it stays at the login prompt.

I notice that screenly-viewer.service won't start:
> Mar 17 17:00:58 raspberrypi systemd[1]: Failed to start Screenly Viewer.
> Mar 17 17:00:58 raspberrypi systemd[1]: Unit screenly-viewer.service entered failed state.
> Mar 17 17:00:58 raspberrypi systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
> Mar 17 17:00:58 raspberrypi systemd[1]: Stopping Screenly Viewer...
> Mar 17 17:00:58 raspberrypi systemd[1]: Starting Screenly Viewer...

In an effort to troubleshoot, I stop the viewer service and run viewer.py from the CLI which yields:
> Traceback (most recent call last):
>   File "screenly/viewer.py", line 343, in <module>
>     main()
>   File "screenly/viewer.py", line 330, in main
>     load_browser(url=url)
>   File "screenly/viewer.py", line 165, in load_browser
>     browser_send(uzbl_rc)
>   File "screenly/viewer.py", line 175, in browser_send
>     if cb(browser.next()):
>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 355, in next
>     self.wait()
>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
>     self._handle_exit_code(self.process.wait())
>   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
>     self.process.stderr
> sh.ErrorReturnCode_1:
> 
>   RAN: '/usr/bin/uzbl-browser --config=- --uri=http://0.0.0.0:8080/splash_page --print-events'
> 
>   STDOUT:
> 
> 
>   STDERR:
> 
> (uzbl-core:8648): Gtk-WARNING **: cannot open display:
> 

Here is my diagnostics.py output:

> {'connectivity': ['http://www.google.com: OK',
>                   'http://www.bbc.co.uk: OK',
>                   'https://www.google.com: OK',
>                   'https://www.bbc.co.uk: OK'],
>  'cpu_info': {'cpu_count': 1,
>               'hardware': 'BCM2708',
>               'model name': 'ARMv6-compatible processor rev 7 (v6l)',
>               'revision': '000e',
>               'serial': '000000007ac5db8c'},
>  'debian_version': '8.0',
>  'git_hash': 'ccd9b46258df13f0e500974d4d5163fb6aaa6b17',
>  'gpu_version': u'48a26a2ae46c497139b3d5a9c8d15485c7b3bfbc (clean) (release)',
>  'hostname': 'raspberrypi',
>  'ifconfig': {'eth0': {2: [{'addr': '192.168.1.27',
>                             'broadcast': '192.168.1.255',
>                             'netmask': '255.255.255.0'}],
>                        10: [{'addr': 'fe80::a449:fb10:b471:4d92%eth0',
>                              'netmask': 'ffff:ffff:ffff:ffff::'}],
>                        17: [{'addr': 'b8:27:eb:c5:db:8c',
>                              'broadcast': 'ff:ff:ff:ff:ff:ff'}]},
>               'lo': {2: [{'addr': '127.0.0.1',
>                           'netmask': '255.0.0.0',
>                           'peer': '127.0.0.1'}],
>                      10: [{'addr': '::1',
>                            'netmask': 'ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff'}],
>                      17: [{'addr': '00:00:00:00:00:00',
>                            'peer': '00:00:00:00:00:00'}]}},
>  'kernel_modules': [u'binfmt_misc',
>                     u'cfg80211',
>                     u'rfkill',
>                     u'snd_bcm2835',
>                     u'snd_pcm',
>                     u'snd_timer',
>                     u'snd',
>                     u'bcm2835_gpiomem',
>                     u'bcm2835_wdt',
>                     u'uio_pdrv_genirq',
>                     u'uio',
>                     u'ipv6'],
>  'loadavg': {'1 min': 4.83, '15 min': 4.1, '5 min': 4.54},
>  'monitor': 'state 0x12000a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive',
>  'ntp_status': '     remote           refid      st t when poll reach   delay   offset  jitter\n==============================================================================\n*time-a.timefreq .NIST.           1 u   29  128  377   38.911    0.412   0.717\n+christensenplac 209.51.161.238   2 u   21  128  317   31.970    1.333   1.739\n-74.120.81.219   45.56.123.24     3 u  212  128  356   51.132   -1.754   0.935\n+195.21.152.161  195.66.241.2     2 u   46  128  377   85.595    6.043   5.131\n',
>  'playlist': [(u'e2a6bd1f03d843d08e6dea9a5e25fee7',
>                u'Big Buck Bunny',
>                u'/home/pi/screenly_assets/e2a6bd1f03d843d08e6dea9a5e25fee7',
>                None,
>                u'2016-12-01 00:00:00',
>                u'2018-12-08 00:00:00',
>                u'N/A',
>                u'video',
>                u'',
>                0,
>                1),
>               (u'1fc6c9735c0143b4bdff4483583b5a71',
>                u'TandT.png',
>                u'/home/pi/screenly_assets/1fc6c9735c0143b4bdff4483583b5a71',
>                None,
>                u'2017-03-07 17:56:00',
>                u'2020-03-14 16:56:00',
>                u'10',
>                u'image',
>                1,
>                0,
>                0),
>               (u'659402d837264392bf63b878b383f66f',
>                u'IMG_5656.JPG',
>                u'/home/pi/screenly_assets/659402d837264392bf63b878b383f66f',
>                None,
>                u'2017-03-07 19:02:00',
>                u'2017-03-10 04:02:00',
>                u'20',
>                u'image',
>                0,
>                0,
>                1)],
>  'uptime': 6143.9,
>  'utc_isodate': '2017-03-17T21:13:59.634299'}

vpetersson, 2017-03-18:
Thanks for sending over the diagnostics. 

My best guess, however, would be that it's related to a failing SD card.

Could you check the system logs to see if you can see any read/write errors in there?

vpetersson, 2017-06-30:
Closing this out due to lack of response.

closed 2017-06-30 by vpetersson