Screenly/Anthias #553: Upload file throught API

thefyfy, 2017-03-31:
I recently create this project : https://github.com/thefyfy/ScreenlyManager
So  I can now manage multiple devices with the same interface because I use a lot of webpages. But for the enhancement of this project, I would like to know if it's possible to upload Image and Video throught API ? 
In this way this project can be used to totaly manage screenly.
Thanks for your amazing job !

vpetersson, 2017-03-31:
I don't know, to be honest. The API was mostly designed for internal usage. It might be possible. Since the API isn't very well documented, your best bet is probably to check the source code in `server.py`.

thefyfy, 2017-04-03:
Thanks I'm going to see that

closed 2017-04-03 by vpetersson