Screenly/Anthias #554: Pi Zero W wireless not recognized

zoidfarb204, 2017-03-31:
The wireless adapter for the pi Zero isn't recognized when using custom img.

vpetersson, 2017-03-31:
Yeah, you would have to use the Raspbian Lite image and install until we get the new image out.

The reason is that firmware in that disk image is older than the Pi Zero W.

closed 2017-03-31 by vpetersson

mckaycr, 2017-11-17:
I have attempted this, but I can't get past installing pip dependencies

vpetersson, 2017-11-17:
Just pull down the [latest disk image](https://github.com/Screenly/screenly-ose/releases/tag/v0.12.1). It should work out-of-the-box on the Pi Zero W. 

We can't troubleshoot the pip dependency errors without logs.

mckaycr, 2017-11-17:
the problem with the latest image is that the pi zero w isn't automatically connecting to wifi, so I have no connection to it.  I have used the wizard to create a network.ini file.

vpetersson, 2017-11-17:
Hmm OK. Any chance you can connect a keyboard to it?

mckaycr, 2017-11-17:
was really trying to avoid buying a hdmi mini adapter and a usb adapter for the zero, but it sounds like that might be my only option for troubleshooting this thing?  It seems like headless installs were so much easier at one time.

vpetersson, 2017-11-17:
> was really trying to avoid buying a hdmi mini adapter and a usb adapter for the zero

Well, without a HDMI mini adapter, Screenly is rather useless on the Pi ;)

> but it sounds like that might be my only option for troubleshooting this thing? 

That rules out USB->Ethernet adapter too I guess. 

The only other thing I can think of is to give resin.io a shot. We got some more content coming, but [this](https://github.com/Screenly/screenly-ose/blob/master/docs/resin.md) should help you get started.

mckaycr, 2017-11-17:
by chance, does your screenly network manager that utilizes the network.ini file have any character limitations in the passphrase?  My passphrase has special characters like $ and @ that I notices in other troubleshooting steps on the RPI3 screenly didn't like.  While using the command line to set the passphrase I had to surround it in single quotes. I was going to try that in the ini file, but I wasn't sure if it would take.

>Well, without a HDMI mini adapter, Screenly is rather useless on the Pi ;)

No you are absolutely right.  What I really meant was, I want to get a proof of concept up and running with what I do have before I go out get everything I will need to go live.  So eventually yes, the hdmi adapter will be bought unless this doesn't pan out, in which the zero will be repurposed for some other headless job.

vpetersson, 2017-11-17:
> by chance, does your screenly network manager that utilizes the network.ini file have any character limitations in the passphrase? 

I don't think so. We're basically just using [Python ConfParser](https://docs.python.org/2/library/configparser.html) to write `/etc/network/interfaces`.

Given the situation, I'd recommend that you give resin.io a shot as they have a different way of doing network configs.

mckaycr, 2017-11-17:
> Given the situation, I'd recommend that you give resin.io a shot as they have a different way of doing network configs.

Thanks, I'll check it out, in the mean time, I somehow got it to work.  57th time is the charm I guess.

mckaycr, 2017-11-17:
resin.io, where have you been all my life.  Thanks for that.  So just for kicks I deployed screenly via resin.  It took, but I'm thinking it doesn't like being run as root?  
Keeps repeating this, and the UI says Server `Error: attempt to write a readonly database`

`17.11.17 14:56:46 (-0500) Nov 17 19:56:46 60bffbe python[963]: Browser loading http://0.0.0.0:8080/splash_page. Running as PID 975.
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: Viewer crashed.
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: Traceback (most recent call last):
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 363, in <module>
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: main()
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 349, in main
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: load_browser(url=url)
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 179, in load_browser
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: browser_send(uzbl_rc)
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 189, in browser_send
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: if cb(browser.next()):
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 355, in next
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self.wait()
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self._handle_exit_code(self.process.wait())
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self.process.stderr
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: ErrorReturnCode_2:
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: RAN: '/usr/bin/uzbl-browser --config=- --uri=http://0.0.0.0:8080/splash_page --print-events'
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: STDOUT:
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: STDERR:
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: mkdir: cannot create directory '/data/.cache': Permission denied
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: could not create /data/.cache/uzbl
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: Traceback (most recent call last):
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 363, in <module>
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: main()
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 349, in main
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: load_browser(url=url)
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 179, in load_browser
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: browser_send(uzbl_rc)
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/home/pi/screenly/viewer.py", line 189, in browser_send
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: if cb(browser.next()):
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 355, in next
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self.wait()
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self._handle_exit_code(self.process.wait())
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: self.process.stderr
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: sh.ErrorReturnCode_2:
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: RAN: '/usr/bin/uzbl-browser --config=- --uri=http://0.0.0.0:8080/splash_page --print-events'
17.11.17 14:56:47 (-0500) Nov 17 19:56:47 60bffbe python[963]: STDOUT:
17.11.17 14:56:48 (-0500) Nov 17 19:56:47 60bffbe python[963]: STDERR:
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe python[963]: mkdir: cannot create directory '/data/.cache': Permission denied
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe python[963]: could not create /data/.cache/uzbl
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: screenly-viewer.service: main process exited, code=exited, status=1/FAILURE
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: Unit screenly-viewer.service entered failed state.
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: Stopping Screenly Viewer...
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: Starting Screenly Viewer...
17.11.17 14:56:48 (-0500) Nov 17 19:56:48 60bffbe systemd[1]: Started Screenly Viewer.`

vpetersson, 2017-11-17:
Smells like a broken SD card to me. Try downloading [SD Formatter](https://www.sdcard.org/downloads/formatter_4/) and do an overwrite format.

mckaycr, 2017-11-17:
>Smells like a broken SD card to me. Try downloading SD Formatter and do an overwrite format.

Hmm... I did use one of my crappier SD cards for this test.  I'll try again with a better one.

mckaycr, 2017-11-17:
No, same thing different card.

vpetersson, 2017-11-17:
ping @antonmolodykh.