Screenly/Anthias #562: Screen stays black after rebooting tv

vanrensav, 2017-05-12:
Hi,

I have an issue using screenly ose.
The raspberry stays on all day and after shop ours they turn off the tv screen only.
When they come back in the morning an start up the tv the raspberry keeps giving black screen.

After they rebooted the raspberry it works again.

What can i do to fix this?

vpetersson, 2017-05-12:
That's strange. You sure they are not rebooting the Raspberry Pi at night with the TV off? Also, how is the monitor powered on/off?

vanrensav, 2017-05-12:
No its on constant power.
Its turned off bij using the remotecontrol when te staff leave te shop.

vanrensav, 2017-05-14:
Any idea? Is it maybe possible to power the raspy by the usb port on the tv? So it starts up when you turn the tv on?

vanrensav, 2017-05-14:
I think something goes wrong at the hdmi port. When raspy is on for a long time and tv is off tv doesnt show any slides. When i reboot the raspy it shows screenly and starts playing. Or is it possible screenly got stuck at some moment?

vanrensav, 2017-05-14:
Looks like the same issue galuwa has in this topic: https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=12396&start=525

edav8458, 2017-06-20:
I had this on our Pi v3 running Screenly OSE. 

The solution for me was to edit the /boot/config.txt  and uncomment/add the following:

`
hdmi_force_hotplug=1 
hdmi_group=1
hdmi_mode=16 
`

vpetersson, 2017-07-01:
@vanrensav 

> Is it maybe possible to power the raspy by the usb port on the tv?

I would *strongly* discourage this, as it will cause a lot of instabilities (both because the TV won't provide enough power and because the random shutdowns will cause the SD card to go corrupt.

In regards to the black screen issue, I would try @edav8458's suggestion and put `hdmi_force_hotplug=1` in your `/boot/config.txt`.

I'm closing this one out as we can't reproduce it. Please let us know if that doesn't solve the issue.

closed 2017-07-01 by vpetersson