Screenly/Anthias #565: No wireless connection after reboot

ppk4, 2017-05-17:
I've started out with installing Raspbian Jessie Lite on to a 2B. 
Done raspi-config and enabled ssh.
Updated wpa_suppliciant.conf file for my wireless connection by entering ssid and psk.
Strangely enough I could not set the proper keyboard (US) layout; it is still showing en_GB. 

Over ssh I've been able to update/upgrade to the latest version of Jessie. Made a backup copy of the SD-card using HDD Raw Copy.  BTW: I did not use any wired network connection.

After reboot I've installed screenly-ose as per instructions  (bash <(curl ... ). This works fine.

Screenly-ose starts up and gives me the management ip address. I've logged in and added some assets. The presentations runs fine. So far, so good.  

However, after rebooting it says 'unable to resolve management ip', but continues to run the presentation (images only). 

As it turns out there is no wifi connection and therefor I can't use ssh. I've rebooted few times without any luck.

iwlist wlan0 scan says 'No scan result'
ifconfig does not show any ip address
lsusb shows that the Ralink RT5370 Wireless Adapter is present

interfaces
`# Generated by Screenly Network Manager |
auto lo  |
     iface lo inet loopback`

_Note: I don't no why the code does not appear as separate lines, so I inserted a | for readability_

wpa_supplicant.conf
`country=NL  |
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev  |
update_config=1  |
network={  |
     ssid="MySSID"  |
     psk="MyKey"  |
}`

The wpa_supplicant does not exist in /var/run/
Value of MySSID and MyKey contain the correct settings for my router.

Entering the ssid and psk in 'interfaces' does not resolve the issue. After rebooting this code has disappeared from the file. After saving the file, I check if the code is there. It was.

When I reinstall jessie and screenly-ose. It works again. After reboot it does not.

I'm rather new in rpi and linux and would appreciate any help. This is my first message in any forum. May be overlooked some things, but I am willing to learn.

Peter-Paul Kolber

vpetersson, 2017-05-18:
Try switching to the development version to see if that resolves the issue. We have some issue that might be causing issues with networking during some certain circumstances.

ppk4, 2017-05-18:
Thank you for the response. Where do I find the development version?

vpetersson, 2017-05-18:
Simply run the upgrade and answer yes to the question if you want to install the development version.

ppk4, 2017-05-18:
I've upgraded to the development version. After rebooting twice, it seem to work. That is to say, if I only include images as assets. 

I'll start a new post on that issue.

ppk4, 2017-05-19:
Today, the wireless connection does not work anymore.  The file network.ini from the wizard I've put in the root of the SD-card shows the ''standard content of the sample version and not my ssid/psk anymore.

The only difference is that I switched back from the development version to the current version. Reason for this is that I couldn't upload images in the production version. Yesterday I only rebooted, but didn't power down.

vpetersson, 2017-05-22:
Can you provide the following output:

 * `cat /etc/network/interfaces`
 * `cat /etc/resolv.conf`
 * `sudo journalctl`
 * `ifconfig -a`

ppk4, 2017-05-22:
At the moment, it works.

closed 2017-05-22 by ppk4