Screenly/Anthias #567: Stops looping after loading webpage

ppk4, 2017-05-18:
I'm running screenly-ose with 3 assets (1920x1080 jpg images). This works fine.

When I include a webpage (add to the active assets), it displays the the webpage. However after showing the next image it stops playing entirely. I've tried a couple of times. The behaviour is consistent.

I've included the log file.
[screenly-ose-log-20170518-pm.txt](https://github.com/wireload/screenly-ose/files/1011276/screenly-ose-log-20170518-pm.txt)

ppk4, 2017-05-22:
Today, I've tested the development version. The behaviour is the same. After displaying the webpage the presentation halts. After pressing CTL-ALT-F2 it starts the sleeping time and displays the next asset. And again it halts. Pressing the keys, it shown after the sleeping time the next asset and halts again.

ppk4, 2017-05-22:
When assets are only images (jpg/png) it works.
When assest are only webpages it works as well.
When assets are a mix of images and webpages it doesn't; it behaves as described in the previous message.
I didn't try videos.

vpetersson, 2017-05-23:
Very good troubleshooting @ppk4. We'll get back to you once we've reproduced the issue.

ppk4, 2017-05-24:
I've have tested a mp4 video. The video itself runs fine. 

Let's summerize the current behaviour.

Assets are only images - works
Asset is only a video - works fine (continues loop)
Assets are only webpages - works

Assets is a mix of images and a video - works

Assets is a mix of images and webpages - stops after displaying a webpage

Four images followed by 5 webpages: display all assets until the second loop needs to start.
It then only displays the first image and then it pauses.
After pressing CTL-ALT twice (and some delay) the next image is shown, it pauses again.
It looks like the duration timer is stopped and waits to get triggered by a key.
Once the first webpage is displayed, the sequence continues till the first image; resulting in a black screen.
Pressing CTL-ALT the first image reappears, but it pauses. And so on after every image.

Five webpages followed by 4 images: 
The 5 webpages are shown, then the system pauses.
Pressing CTL-ALT shows the first image and pauses.
Now, the behaviour is the same as of the second cycle in the previous example.

Assets is a mix of a video and webpages - stops after displaying the video
Pressing any key on the keyboard will proceed the playlist until the next time the video is shown.
The position of the video in the playlist does not change the behaviour.
Every time, the presentation stops after displaying the video. It is consistent. 

Assets is a combination of images, video and webpages.
Behaviour is inconsistent. Transistion do not work until you press a key once or twice.

Conclusion: If a webpage is not followed by a webpage, the system pauses and waits for a trigger.

Regards,
Peter-Paul

Raspberry 2B
Linux screenly 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv 
Note: I do not know how to print the version of screenly-ose (but it is the latest one).

ppk4, 2017-05-24:
Additional comment:
If I turned off the webpages and leave the images and video in the playlist, the behaviour does not change! Only until I do 'Save settings', the images and video are displayed flawlessly.

ppk4, 2017-05-24:
So far, I have tested webpages from only one website. The website of our tennis club. In our clubhome we have a large screen on which we want to show news, images, video and webpages.

I have tried a few more things like taking sources from a few other websites. To my surprise, all those worked! So the problem seems to be related to one particular website (http://www.mltcready.nl ; built by a professional company). Unfortunately the one I want to use.

ppk4, 2017-05-24:
Despite all comments, I really love Screenly running on a Raspberry Pi.

vpetersson, 2017-06-30:
Adding this to the next sprint for further investigation.

antonmolodykh, 2017-08-08:
I could not reproduce the stoppage of asset playing. However, when there are web pages in the playlist, there are really small pauses with a black screen. This is the case when we are dealing with heavy or poorly optimized sites

vpetersson, 2017-08-16:
Closing out as we're not able to reproduce.

closed 2017-08-16 by vpetersson