Screenly/Anthias #568: OSE development version does not allow upload of images

ppk4, 2017-05-18:
Today's development version does not allow me to upload images. 
Note: The dialog screen is different than in the 'normal' version.

Reverting back to the previous version worked fine.

antonmolodykh, 2017-05-19:
you're right, the dialog screen now looks different.
what exactly happens when you try to upload images?

ppk4, 2017-05-19:
Drag-and-drop does not work.  It opens the image in my browser. Dialog disappears.
Upload image does not work. No dialog. It doesn't allow me to attach a file.

vpetersson, 2017-05-19:
@ppk4 what browser/version/os?

ppk4, 2017-05-19:
Windows 7 SP1 and Firefox 53.0.2 (64-bit)

antonmolodykh, 2017-05-22:
@ppk4 Today I've tried to use win 7, firefox 53.0.2, 64-bit. I haven't managed to reproduce the problem

ppk4, 2017-05-22:
After the previous meesage, I switched back to the official release (not the development version).

Today, I reinstalled the development version. To my surprise the upload of image worked.

[image]
[image]

According to the dialog screens, you do not know which image has been uploaded. You only get a message 'Loading is complete'.  However it worked.

If you want I can still upload the four files,

closed 2017-05-31 by ppk4