Screenly/Anthias #575: webpage not accessible after update

thhartung, 2017-06-16:
Last week I updated two of our screenly clients.
I was able to access the webinterfaces and upload content like before.
Today after a rebbot both clients webinterfaces are not responding anymore, ssh login is still possible.
Quick port scan showed no open ports except the ssh port anymore

vpetersson, 2017-06-16:
Try upgrading to the development version and let us know if that resolves the issue.

thhartung, 2017-06-16:
switched to the development branch, same problem

vpetersson, 2017-06-16:
What do you get from `journalctl --since "24 hours ago"`

thhartung, 2017-06-16:
Stange enough mothing, doenst semm to be installed
I started the installation once again, here is she output, there seem to be some failures

[Neues Textdokument.txt](https://github.com/Screenly/screenly-ose/files/1080460/Neues.Textdokument.txt)

vpetersson, 2017-06-16:
Hmm odd. Well, you're on Wheezy. You probably want to upgrade to Jessie anyways, as Wheezy is getting very old by now. I suggest you fetch the latest disk image and use that instead.

closed 2017-06-16 by vpetersson