Screenly/Anthias #576: Add Asset image throws Uncaught TypeError Cannot read property 'size' of undefined

vthyng, 2017-06-19:
Happens when trying to add an image asset clicking the Upload File button.  I get this on a fresh install from Raspbian and the following configuration changes:
enabled screenly SSL
Auth credentials set

Adding a URL works.
The code indicates it is trying to access a file object that does not exist.  It seems like it should not be checking for a file size before I have specified a file.

```
Uncaught TypeError: Cannot read property 'size' of undefined
    at e.(anonymous function).(anonymous function)._chunkedUpload (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery.fileupload.js:540:26)
    at e.(anonymous function).(anonymous function)._chunkedUpload (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery-ui-1.10.1.custom.min.js:6:5077)
    at e.(anonymous function).(anonymous function)._initDataSettings (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery.fileupload.js:425:27)
    at e.(anonymous function).(anonymous function)._initDataSettings (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery-ui-1.10.1.custom.min.js:6:5077)
    at e.(anonymous function).(anonymous function)._getAJAXSettings (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery.fileupload.js:494:18)
    at e.(anonymous function).(anonymous function)._getAJAXSettings (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery-ui-1.10.1.custom.min.js:6:5077)
    at e.(anonymous function).(anonymous function)._onSend (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery.fileupload.js:703:32)
    at e.(anonymous function).(anonymous function)._onSend (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery-ui-1.10.1.custom.min.js:6:5077)
    at Object.<anonymous> (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery.fileupload.js:1142:42)
    at c (https://status-pi.prod.sea1.west.isilon.com/static/js/jquery-1.9.1.min.js:3:7857)
```

vpetersson, 2017-06-20:
@antonmolodykh can you see if you can reproduce this?

vpetersson, 2017-08-03:
@vthyng Can you try this with the latest development version?

Closing out the issue until we can reproduce it.

closed 2017-08-03 by vpetersson