Screenly/Anthias #592: No assets shown and can't add new assets (after update)

Aokuro, 2017-07-11:
Hi,

after updating screenly to the latest version I encountered some grave errors.

When I connect to the webfrontend there are no assets shown. The strange part is, that all the previous available assets are still shown on the Raspberry Pi as a slideshow, but there is no chance for me to add, edit or delete any asset (except for the delete part via terminal ofc :) )

I already tried to setup a completely new Screenly which didn't really help. I backed up the old Settings via frontend and uploaded them onto the new Screenly. For a few minutes the assets were up and running again, but after waiting for a bit they disappeared again.

When I try to add a new asset via the button I only get redirected to: ip.ip.ip.ip:8080/#

Here a screen of the frontend and everything I can see: 

[image]

I'll try using an older version of Screenly for now, hope somebody can solve this issue or knows something more. Thanks in advance.

Greetings
Aokuro

vpetersson, 2017-07-11:
@antonmolodykh Any thoughts here?

vpetersson, 2017-07-11:
@Aokuro can you please post a the logs under the 'System info' section.

kpjuergensen, 2017-08-27:
Hi guys,
have the same issue. The problem occurs with Chrome, but it works with Edge or Safari. But sometimes loading up a video or picture got the same error: "The operation failed. Please reload the page and try again."

vpetersson, 2017-08-27:
@kpjuergensen can you please post the logs under the 'System info' section.

kpjuergensen, 2017-08-27:
Hi @vpetersson,

that are the logs after trying to add an URL asset:

> * screenly-viewer.service - Screenly Viewer
> Loaded: loaded (/etc/systemd/system/screenly-viewer.service; enabled)
> Active: active (running) since So 2017-08-27 23:28:38 CEST; 11min ago
> Process: 724 ExecStartPost=/bin/rm -f /tmp/screenly_html/* (code=exited, status=0/SUCCESS)
> Process: 722 ExecStartPost=/bin/rm -f /tmp/uzbl_* (code=exited, status=0/SUCCESS)
> Process: 718 ExecStartPre=/usr/bin/xset s noblank (code=exited, status=0/SUCCESS)
> Process: 712 ExecStartPre=/usr/bin/xset -dpms (code=exited, status=0/SUCCESS)
> Process: 696 ExecStartPre=/usr/bin/xset s off (code=exited, status=0/SUCCESS)
> Main PID: 721 (python)
> CGroup: /system.slice/screenly-viewer.service
> \|-721 /usr/bin/python /home/pi/screenly/viewer.py

At _So 2017-08-27 23:28:38 CEST;_ was the Pi rebooted. After this, I tried to add something. The result is shown in the IMG. The error on top and the buttons are both greyed out. Only ESC helps to get back. It seems as if no logs are written.

[image]

IMG of the log. Debug loggin is on.
[image]

Hope this will help.

antonmolodykh, 2017-09-12:
I think that problem is in browser cache 
The javascript part has been seriously changed, and if my guess is correct, then this explains the problem, and also explains the correct operation in other browsers. @kpjuergensen  please try again and after the update, clear the cache in chrome

vpetersson, 2017-09-26:
Closing due to lack of response.

closed 2017-09-26 by vpetersson

Ruko2010, 2019-02-05:
i know this is an old issue, but I ran into the same problem. I updated screenly from console and after the reboot I get the same message.
I tryed to clear the browser cache and even used a browser never visited the screenly page before, but the error persists. Here is my Systemlog:
[image]

Ruko2010, 2019-02-05:
Nevermind. I had an error on the update, and reinstalling the update solved the problem. It is now working as it was before. Thanks for the great work.