Screenly/Anthias #593: Raspberry Pi 3: Wifi Stops Working After First Boot

KU7PDX, 2017-07-17:
I've been working on trying to track down a solution, but haven't had any luck.

After imaging a new SD card (using the provided Screenly image or building from a Raspbian Lite image) for my RPi 3 and placing the following files onto it, Screenly will boot and show my management IP and everything looks great.

After I reboot or shutdown (and later restart), the RPi 3 won't attempt to connect on wifi and I receive an error stating "Unable to resolve management IP."

The problem is consistent and here are the steps to re-create:

Image 2017-07-05-raspbian-jessie-lite.img to an SD card. Immediately copy the attached config.txt, ssh, and wpa_supplicant.conf (listed as a TXT file for Github) files to SD card boot directory.

[config.txt](https://github.com/Screenly/screenly-ose/files/1151635/config.txt)
[wpa_supplicant.conf.txt](https://github.com/Screenly/screenly-ose/files/1151636/wpa_supplicant.conf.txt)

I boot the Pi up and wait for the IP to come up on my monitor, then I SSH into it. After connecting, I run the following:

sudo apt-get update && sudo apt-get dist-upgrade
(let everything upgrade)

curl https://get.pimoroni.com/onoffshim | bash
(I'm also using a Pimoroni OnOff SHIM power switch, pretty awesome stuff)

bash <(curl -sL https://www.screenly.io/install-ose.sh)
(I don't run the development branch)

After the install completes, I run a sudo reboot to restart the Pi. Everything comes up fine, so I SSH back into the Pi and do either a sudo reboot again or sudo poweroff and then power cycle the Pi.

On the next boot, Screenly will report that it is unable to resolve the management IP. Even copying the wpa_supplicant.conf back to the SD card and booting again doesn't resolve the issue.

Any suggestions or tips would be greatly appreciated, thanks!

vpetersson, 2017-07-17:
So the WiFi entirely breaks, or are you still able to SSH in, but it doesn't report the right IP?

In any case, could you provide the following:

* `cat /etc/networking/interfaces`
*  `sudo journalctl`
* `ifconfig -a`

(if the network breaks, you can of course not SSH in, but maybe you can use an ethernet cable as a temporary remedy.)

KU7PDX, 2017-07-17:
There is no wireless connectivity, so I am unable to SSH in after installing Screenly and rebooting.

I don't have an available ethernet jack, but I'll try finding a USB keyboard to connect locally later tonight.

KU7PDX, 2017-07-18:
Per your request! It looks like for some reason Screenly booted the wlan0 interface after rebooting.

[journalctl.txt](https://github.com/Screenly/screenly-ose/files/1154630/journalctl.txt)
[ifconfig.txt](https://github.com/Screenly/screenly-ose/files/1154628/ifconfig.txt)
[interfaces.txt](https://github.com/Screenly/screenly-ose/files/1154629/interfaces.txt)

MrHae, 2017-07-18:
just insert the sd card into your notebook or pc and edit the network.ini_sample
Rename it to network.ini and configure your wireless connection here
after booting screenly overwrites /etc/network/interfaces

vpetersson, 2017-07-19:
@KU7PDX Looking at your config, I can confirm that you must either be running on an older disk image, or have a `/boot/network.ini` file present.

The real issue is that the `/etc/interfaces` file gets populated from `/boot/network.ini`, which doesn't have any configuration for `wifi0`. 

Honestly, the quickest solution is likely to use [wizard.screenly.io](https://wizard.screenly.io/) to configure your network and follow the instructions there.

KU7PDX, 2017-07-24:
So, after more troubleshooting today, I figured out that after installing Screenly (via bash <(curl -sL https://www.screenly.io/install-ose.sh)), it would reset all of my network settings and ignore changes until I generated a network.ini file and placed it into the boot folder.

This should probably be documented a bit better, as there isn't any indication that the network settings would be reset.

Thanks for everyone's help!

closed 2017-07-24 by KU7PDX