Screenly/Anthias #595: InsecureRequestWarning

JacoFourie1, 2017-07-24:
Hi. When I try to load a site to show a dashboard I get this error

`Jul 24 11:40:09 raspberrypi python[819]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
Jul 24 11:40:09 raspberrypi python[819]: InsecureRequestWarning)
Jul 24 11:40:09 raspberrypi python[819]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
Jul 24 11:40:09 raspberrypi python[819]: InsecureRequestWarning)
Jul 24 11:40:09 raspberrypi python[819]: Traceback (most recent call last):
Jul 24 11:40:09 raspberrypi python[819]: File "/home/pi/screenly/server.py", line 194, in api_view
Jul 24 11:40:09 raspberrypi python[819]: return make_json_response(view(*args, **kwargs))
Jul 24 11:40:09 raspberrypi python[819]: File "/home/pi/screenly/server.py", line 230, in add_asset
Jul 24 11:40:09 raspberrypi python[819]: raise Exception("Could not retrieve file. Check the asset URL.")
Jul 24 11:40:09 raspberrypi python[819]: Exception: Could not retrieve file. Check the asset URL.
`
It seems it does not like the certificate. I did have a look at the documentation at https://urllib3.readthedocs.org/en/latest/security.html but I am not sure where to add the line of code to switch off the warning.

JacoFourie1, 2017-07-25:
I am using the pre build image that I downloaded from the site. It is Jessie.

vpetersson, 2017-07-25:
Can you provide a sample URL where you get this error?

vpetersson, 2017-11-21:
Closing out due to lack of response.

closed 2017-11-21 by vpetersson

ealmonte32, 2019-04-01:
@JacoFourie1 
To stop getting the warning make the change on the `/home/pi/screenly/lib/utils.py` file like this:

[image]