Screenly/Anthias #599: Localise date format

daniep01, 2017-07-25:
Hi,
It would be better if the displayed date format in the schedule could be localised (dd/mm/yyyy) or (mm/dd/yyyy) based on a user setting. I'm happy to work on this but would like some pointers from more experienced contributors - 
* is there a reason this hasn't/can't be done
* where in the code to start looking

Thanks.

vpetersson, 2017-07-26:
@daniep01 If I'm not mistaken, there should already be support for this. I guess the main consideration is that we render them to native dates in the backend.

vpetersson, 2017-07-26:
@antonmolodykh do you have any pointers here?

antonmolodykh, 2017-07-27:
If we are talking about a new parameter in the settings, then it's not difficult:
1. you need to add a new parameter in `settings.py` and `screenly.conf`
2. Pass this parameter to the template (function `template` in `server.py`)
3. Add a new parameter to the `settings.haml` template
3. In `screenly-ose.coffee` make a condition which in the format will be selected according to the passed parameter.

If I'm not mistaken, now we don't have the ability to detect a location

vpetersson, 2017-07-27:
@antonmolodykh Shouldn't we be able to detect that with JS [relatively easy](https://stackoverflow.com/questions/673905/best-way-to-determine-users-locale-within-browser)?

antonmolodykh, 2017-07-27:
@vpetersson As I understand it is about parsing the "Accept-Language" header. If I understand correctly, then parsing involves processing on the server side, and even in this case it's just an attempt to find out the locale, nothing more. We still have to provide a choice in case of an error. It seems to me easier to provide a choice of format through settings

vpetersson, 2017-07-27:
Perhaps we can save ourselves the effort and use [this feature](https://bottle-utils.readthedocs.io/en/latest/i18n.html) in Bottle?

antonmolodykh, 2017-07-27:
If I understand correctly, this module provides translation assistance, I'm not sure that I understand how this can help us. In any case, the essence of this task is not related to the language: for example, in the US and in the UK different date formats are used.

vpetersson, 2017-07-27:
I just saw this piece:

```
<p>{{ _('Current time') }}: {{ time }}</p>
```

I *think* that provides the localized time, but i could be wrong.

antonmolodykh, 2017-07-27:
It seems to me to detected a local on the server side is not a good idea. If I understand correctly, for users who downloaded the Screenly image and did not configure the locale on their own, the data will not be correct. Am I mistaken?

vpetersson, 2017-07-27:
@antonmolodykh i think you misunderstand the feature. The locale is derived from the user. Not from the server. It's just processed server side such that the correct content, which is can be conditional, can be served to the user.

daniep01, 2017-08-01:
Does it make sense to combine the date format setting with the existing 12/24 hour format setting so that US style dates have 12h time and UK style dates are 24h time? Or do they need to be separate?

vpetersson, 2017-08-03:
@daniep01 I'm not sure I'm following. In my opinion, the ideal solution would look like this:

* When the user navigates to the interface, the appropriate format is picked using the browser's locale. 
* We would then add a setting somewhere in the interface where the user can set the desired format should they not want the auto-detected format.
* The back-end (i.e. how it's stored in the database and handled in the server) would remain the same.

I would imagine there are a few different ways of implement this with various frameworks/approaches, but that would be what I would recommend.

daniep01, 2017-08-03:
Hi, yes to all of the above. I'm just wondering if the 12/24 time format should also be part of the locale.

vpetersson, 2017-08-03:
Yeah, I'd say 12h vs 24h should be part of the locale. I think that is what people would expect.

closed 2018-10-19 by stale[bot]