Screenly/Anthias #609: Inherit network configuration from NOOBS

vpetersson, 2017-08-04:
It appears as there is a mechanism in NOOBS for configuring the network. Currently we do not take that into account, but we should look into how this is done and how we can tap into it.

rdcole, 2017-08-13:
When Running WiFi, If I boot NOOBS and select screenly-ose install it installs fine then starts with no problem.  The Pi will do a Warm boot and the Network settings remain and screenly runs as expected. 
However after a Power Cycle Cold Boot the Network does not Restart so screenly reports it cant resolve the management IP address and screenly has no access to content.

tomscragg, 2017-08-21:
Hi Viktor,

I installed today, Raspbian, Windows IoT and Screenly today and initially booted into Win IoT. having messed with that for a while first, I unplugged and restarted the system into Screenly. this did initially connect to the network and I got an IP address. upon restart I had the same issue as before. I thought it worth a mention as it maybe to do with the first time the software starts?

vpetersson, 2017-08-21:
Interesting. Yes, it must be something NOOBS does differently between different boot types. I've added this to the upcoming sprint, so we will try to reproduce and solve this.

tomscragg, 2017-10-05:
Hi Viktor,

any updates on this? I'm struggling to install the product another way and would be great to get this working :-)

vpetersson, 2017-10-05:
Hi @tomscragg - yes @antonmolodykh is working on this. I think he made some progress today.

vpetersson, 2017-10-10:
Just to summarize. Instead of inheriting the network configuration, we have disabled the Screenly Network Manager if NOOBS is detected to avoid conflicts.

closed 2017-10-10 by vpetersson