Screenly/Anthias #621: 4k TVs configuration

kpjuergensen, 2017-08-27:
Hi,
is it possible to configure the resolution settings for 4K TVs? I Read about that the Pi is able to run with 4K resolution (3840x2160). Are there any working and tested settings?

If tried this in the config.txt:

`hdmi_group=2`
`hdmi_mode=87`
`hdmi_cvt 3840 2160 15`
`max_framebuffer_width=3840`
`max_framebuffer_height=2160`
`hdmi_pixel_freq_limit=400000000`

But the problem is, the TV tells me:"No supportet resolution. Pleas change". The data sheet of the TV indicates that the resolution is supported.

Maybe someone has an idea.

Thanks.

vpetersson, 2017-08-28:
While I have heard rumors that it is theoretically possible to get the Pi to run in 4k, it isn't an officially supported resolution. Moreover, the Raspberry Pi isn't a very powerful device. You might be able to show images in 4k, but you won't be able to display videos in 4k native.

vpetersson, 2017-10-09:
Closing out this issue as it's not a Screenly issue per se.

closed 2017-10-09 by vpetersson