Screenly/Anthias #624: "No active network connection found" on Raspbian stretch

anayrat, 2017-09-02:
Hello,

I tried a fresh install on rasbian stretch. After the installation I got this error "No active network connection found"

```
sept. 02 23:14:09 raspberrypi python[529]: Traceback (most recent call last):
sept. 02 23:14:09 raspberrypi python[529]:   File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
sept. 02 23:14:09 raspberrypi python[529]:     return route.call(**args)
sept. 02 23:14:09 raspberrypi python[529]:   File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1732, in wrapper
sept. 02 23:14:09 raspberrypi python[529]:     rv = callback(*a, **ka)
sept. 02 23:14:09 raspberrypi python[529]:   File "/home/pi/screenly/server.py", line 316, in viewIndex
sept. 02 23:14:09 raspberrypi python[529]:     my_ip = get_node_ip()
sept. 02 23:14:09 raspberrypi python[529]:   File "/home/pi/screenly/lib/utils.py", line 83, in get_node_ip
sept. 02 23:14:09 raspberrypi python[529]:     raise Exception("No active network connection found.")
sept. 02 23:14:09 raspberrypi python[529]: Exception: No active network connection found.
```

My Rpi has network connection with ethernet cable:

```
 ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enxb827eb6c50a7: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:6c:50:a7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.19/24 brd 192.168.1.255 scope global enxb827eb6c50a7
       valid_lft forever preferred_lft forever
    inet6 2a01:cb14:2e8:900:34a1:8d59:3e8f:5f43/64 scope global dynamic
       valid_lft 852sec preferred_lft 252sec
    inet6 fe80::740:15d6:d242:365e/64 scope link
       valid_lft forever preferred_lft forever
```

boschie, 2017-09-04:
In #625 @vpetersson tells to use Raspbian Jessie. Anyone knows where to download a trusted Raspbian-lite Jessie? Because I can not find it anymore on the official raspberrypi.org site.

boschie

vpetersson, 2017-09-04:
@boschie Yeah, looks like you're right. I'd suggest either you use NOOBS (you can probably find it there) or our disk image (which you can find [here](https://github.com/Screenly/screenly-ose/releases)).

boschie, 2017-09-05:
Just found it! :)) 

Most files can be retrieved from https://downloads.raspberrypi.org/

john-arvid, 2017-09-08:
Maybe add something more to the README than: "The tl;dr for installing Screenly OSE on Raspbian Jessie is" ?

Frustrating to spend time troubleshooting when this is a known issue.

jarrettwold, 2017-09-09:
What changed between Jessie and Stretch.  
https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/

closed 2017-09-21 by vpetersson