Screenly/Anthias #625: Raspberry Pi 3 Stretch, Networking not working

trbarrettjr, 2017-09-03:
I am utilizing the Stretch Raspbian Lite Image.  Attempted to load the software as described.  When I would reboot after the completion it would not connect back to the network via ethernet cable.

Here are the following logs.

debug.log:
`Aug 16 01:22:13 raspberrypi thd[321]: Found socket passed from systemd
Aug 16 01:22:13 raspberrypi kernel: [    0.000000] On node 0 totalpages: 255488
Aug 16 01:22:13 raspberrypi kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c6eec0, node_mem_map bd336000
Aug 16 01:22:13 raspberrypi kernel: [    0.000000]   Normal zone: 2246 pages used for memmap
Aug 16 01:22:13 raspberrypi kernel: [    0.000000]   Normal zone: 0 pages reserved
Aug 16 01:22:13 raspberrypi kernel: [    0.000000]   Normal zone: 255488 pages, LIFO batch:31
Aug 16 01:22:13 raspberrypi kernel: [    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Aug 16 01:22:13 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Aug 16 01:22:13 raspberrypi kernel: [    0.750953] dwc_otg: Microframe scheduler enabled
Aug 16 01:22:13 raspberrypi kernel: [    0.812767] dwc_otg: FIQ enabled
Aug 16 01:22:13 raspberrypi kernel: [    0.812771] dwc_otg: NAK holdoff enabled
Aug 16 01:22:13 raspberrypi kernel: [    0.812775] dwc_otg: FIQ split-transaction FSM enabled
Aug 16 01:22:13 raspberrypi kernel: [    0.812790] Module dwc_common_port init
Sep  2 20:42:49 raspberrypi kernel: [    0.000000] On node 0 totalpages: 255488
Sep  2 20:42:49 raspberrypi kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c6eec0, node_mem_map bd336000
Sep  2 20:42:49 raspberrypi kernel: [    0.000000]   Normal zone: 2246 pages used for memmap
Sep  2 20:42:49 raspberrypi kernel: [    0.000000]   Normal zone: 0 pages reserved
Sep  2 20:42:49 raspberrypi kernel: [    0.000000]   Normal zone: 255488 pages, LIFO batch:31
Sep  2 20:42:49 raspberrypi kernel: [    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Sep  2 20:42:49 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Sep  2 20:42:49 raspberrypi kernel: [    0.757218] dwc_otg: Microframe scheduler enabled
Sep  2 20:42:49 raspberrypi kernel: [    0.799866] dwc_otg: FIQ enabled
Sep  2 20:42:49 raspberrypi kernel: [    0.799870] dwc_otg: NAK holdoff enabled
Sep  2 20:42:49 raspberrypi kernel: [    0.799875] dwc_otg: FIQ split-transaction FSM enabled
Sep  2 20:42:49 raspberrypi kernel: [    0.799888] Module dwc_common_port init
Sep  2 20:42:49 raspberrypi thd[366]: Found socket passed from systemd
Sep  2 21:25:47 raspberrypi kernel: [    0.000000] On node 0 totalpages: 255488
Sep  2 21:25:47 raspberrypi kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c6eec0, node_mem_map bd336000
Sep  2 21:25:47 raspberrypi kernel: [    0.000000]   Normal zone: 2246 pages used for memmap
Sep  2 21:25:47 raspberrypi kernel: [    0.000000]   Normal zone: 0 pages reserved
Sep  2 21:25:47 raspberrypi kernel: [    0.000000]   Normal zone: 255488 pages, LIFO batch:31
Sep  2 21:25:47 raspberrypi kernel: [    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Sep  2 21:25:47 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Sep  2 21:25:47 raspberrypi kernel: [    0.797357] dwc_otg: Microframe scheduler enabled
Sep  2 21:25:47 raspberrypi kernel: [    0.799489] dwc_otg: FIQ enabled
Sep  2 21:25:47 raspberrypi kernel: [    0.799494] dwc_otg: NAK holdoff enabled
Sep  2 21:25:47 raspberrypi kernel: [    0.799498] dwc_otg: FIQ split-transaction FSM enabled
Sep  2 21:25:47 raspberrypi kernel: [    0.799511] Module dwc_common_port init
Sep  2 21:25:47 raspberrypi thd[364]: Found socket passed from systemd
Sep  2 21:29:29 raspberrypi kernel: [    0.000000] On node 0 totalpages: 255488
Sep  2 21:29:29 raspberrypi kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c6eec0, node_mem_map bd336000
Sep  2 21:29:29 raspberrypi kernel: [    0.000000]   Normal zone: 2246 pages used for memmap
Sep  2 21:29:29 raspberrypi kernel: [    0.000000]   Normal zone: 0 pages reserved
Sep  2 21:29:29 raspberrypi kernel: [    0.000000]   Normal zone: 255488 pages, LIFO batch:31
Sep  2 21:29:29 raspberrypi kernel: [    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Sep  2 21:29:29 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Sep  2 21:29:29 raspberrypi thd[329]: Found socket passed from systemd
Sep  2 21:29:29 raspberrypi kernel: [    0.797149] dwc_otg: Microframe scheduler enabled
Sep  2 21:29:29 raspberrypi kernel: [    0.799199] dwc_otg: FIQ enabled
Sep  2 21:29:29 raspberrypi kernel: [    0.799204] dwc_otg: NAK holdoff enabled
Sep  2 21:29:29 raspberrypi kernel: [    0.799208] dwc_otg: FIQ split-transaction FSM enabled
Sep  2 21:29:29 raspberrypi kernel: [    0.799221] Module dwc_common_port init
Sep  2 21:31:42 raspberrypi thd[317]: Found socket passed from systemd
Sep  2 21:31:42 raspberrypi kernel: [    0.000000] On node 0 totalpages: 255488
Sep  2 21:31:42 raspberrypi kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c6eec0, node_mem_map bd336000
Sep  2 21:31:42 raspberrypi kernel: [    0.000000]   Normal zone: 2246 pages used for memmap
Sep  2 21:31:42 raspberrypi kernel: [    0.000000]   Normal zone: 0 pages reserved
Sep  2 21:31:42 raspberrypi kernel: [    0.000000]   Normal zone: 255488 pages, LIFO batch:31
Sep  2 21:31:42 raspberrypi kernel: [    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Sep  2 21:31:42 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Sep  2 21:31:42 raspberrypi kernel: [    0.797192] dwc_otg: Microframe scheduler enabled
Sep  2 21:31:42 raspberrypi kernel: [    0.799275] dwc_otg: FIQ enabled
Sep  2 21:31:42 raspberrypi kernel: [    0.799280] dwc_otg: NAK holdoff enabled
Sep  2 21:31:42 raspberrypi kernel: [    0.799284] dwc_otg: FIQ split-transaction FSM enabled
Sep  2 21:31:42 raspberrypi kernel: [    0.799297] Module dwc_common_port init
[... 339 lines cut ...]
CPU min MHz:           600.0000
BogoMIPS:              38.40
Flags:                 half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
Module                  Size  Used by
cmac                    3239  1
bnep                   12051  2
hci_uart               20020  1
btbcm                   7916  1 hci_uart
bluetooth             365511  24 hci_uart,bnep,btbcm
evdev                  12423  2
brcmfmac              223048  0
brcmutil                9092  1 brcmfmac
cfg80211              543091  1 brcmfmac
rfkill                 20851  4 bluetooth,cfg80211
snd_bcm2835            24427  0
snd_pcm                98501  1 snd_bcm2835
snd_timer              23968  1 snd_pcm
snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
fixed                   3285  0
ip_tables              13161  0
x_tables               20578  1 ip_tables
ipv6                  409035  24`

anayrat, 2017-09-03:
It seems duplicate of #624

vpetersson, 2017-09-03:
Yup. Closing out this one. For now, use Jessie until we've figured this out.

closed 2017-09-03 by vpetersson