Screenly/Anthias #627: Can't update from NOOBS

anayrat, 2017-09-05:
Hello,

After a fresh install with NOOBS I tried to update and it failed during installation of pyzmq.

If I try manualy I got these errors : 
```
    Settings obtained from pkg-config: {'libraries': [u'zmq'], 'library_dirs': [], 'include_dirs': []}
    {'libraries': [u'zmq'], 'extra_link_args': [], 'runtime_library_dirs': [], 'library_dirs': [], 'include_dirs': []}
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -c build/temp.linux-armv7l-2.7/scratch/check_sys_un.c -o build/temp.linux-armv7l-2.7/scratch/check_sys_un.o
    In file included from /usr/include/stdio.h:27:0,
                     from build/temp.linux-armv7l-2.7/scratch/check_sys_un.c:1:
    /usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory
     #  include <sys/cdefs.h>
                             ^
    compilation terminated.
    Warning: No sys/un.h, IPC_PATH_MAX_LEN will be undefined: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```
I didn't find which package is missing :(

vpetersson, 2017-09-05:
That's interesting. 

My guess would be `python-dev` or `build-essentials`. @antonmolodykh any ideas?

vpetersson, 2017-10-09:
@anayrat I believe this should have been fixed by now. Closing out. Let us know if it doesn't work in the development branch.

closed 2017-10-09 by vpetersson

aolszowka, 2017-10-09:
I just updated this past weekend (and was running into this bug) I reran the update last night using the Development Branch and it worked (but was terribly slow; but its an Original RPi B+ so its to be expected). I'd close as resolved.