Screenly/Anthias #630: Screenly Cast for WordPress not Working

zafersn, 2017-09-09:
I run a WordPress page on my wamp server on my computer.
https://github.com/screenly/screenly-cast-for-wordpress. The system does not work right though I have correctly applied all the steps you have explained. (Only the image does not load, I can read the writings of the image)

I uploaded the link like this:

http://192.168.1.19/wordpress/4866902-blue-picture/?srly

vpetersson, 2017-09-13:
@petermonte any ideas?

petermonte, 2017-09-13:
@zafersn is you're link `http://192.168.1.19/wordpress/4866902-blue-picture/?srly` suppose to be public and running correctly without the `?srly` var?

zafersn, 2017-09-13:
hi @petermonte  @vpetersson  .   The url you have seen is correct. When I connect from the computer or from the phone, the page opens correctly. the page opens in raspberry but the picture does not load. I can read the writings of the image correctly.

petermonte, 2017-09-13:
@zafersn thanks for the quick response. Are you able to access the browser (Chrome, Safari, Firefox...) developer tools in order check what is being printed out on markup?

zafersn, 2017-09-16:
@petermonte  sorry. I'm late. When I look at the console tab from the developer tools of my browser, there is something like the image below.

http://stackcuriosity.com/wp-content/uploads/2017/09/err_screenly.jpg

image on raspberry screen:

http://stackcuriosity.com/wp-content/uploads/2017/09/I.jpg

petermonte, 2017-09-18:
@zafersn thanks for the feedback.

What I was trying to understand is how are the files being called on markup. For that I would need you to print screen the "elements" tab on developers console a focus one of the assets (can bee css or an image). Can you make sure you have all files correctly from the plugin?

On the other hand I'm noticing is a forbidden status from server. This might be a totally different issue not related to missing assets but rather a domain conflict.

Thanks

vpetersson, 2017-09-26:
Migrated [here](https://github.com/Screenly/Screenly-Cast-for-WordPress/issues/13) as it isn't a Screenly OSE issue.

closed 2017-09-26 by vpetersson