Screenly/Anthias #632: Add option for default asset display length

skiracerdude, 2017-09-11:
I would like to suggest a feature which would set a default asset display length. 

For example, I'd like the default display time for an asset to be set at 7 days, so that every time I add an Asset, it will appear for 7 days.

vpetersson, 2017-09-13:
We're happy to accept a pull request for this, as it seems to make sense.

daniep01, 2017-10-18:
I have added this to my system and it seems to be working, what should the new option be called? I'm using 'period' but that might not be best. We need something that won't be confused with the existing 'duration' setting.
@skiracerdude @vpetersson

skiracerdude, 2017-10-19:
How about 'Default Asset Campaign Length' ? I know its kinda long, but it's decently clear. I think it would also help if we add the units being used after the input fields.
Eg. 'Default Duration <input field> (seconds)' 

I think specifying units will also help with the confusion between the two. @danielp01

vpetersson, 2017-10-19:
I agree with @skiracerdude - 'Default duration' makes sense. Having 'Seconds' would be nice too but it would make it too long. Perhaps we could add a `glyphicon glyphicon-question-sign` and then an explainer text that it's in seconds when hovering over it?

daniep01, 2017-10-19:
- 'Default duration (seconds)'
- 'Default streaming duration (seconds)'
- 'Default schedule period (days)'
maybe? To me campaign is too advertising specific, not all users will understand this.

vpetersson, 2017-10-19:
@daniep01 Did you break them out for different asset types? If not I would go with the 'Default duration' and then a helper.

daniep01, 2017-10-19:
think the request is for the default time period (in days) between the start date and end date to be configurable. This isn't the same as the default time that the asset is displayed (in seconds). 
The split between streaming asset and uploaded asset already exists.

vpetersson, 2017-10-19:
Oh, yeah you're right. My bad.

In that case, I think we need a proper, like [this](https://github.com/koss-lebedev/bootstrap-duration-picker) for it to work well.

skiracerdude, 2017-10-19:
@vpetersson Yeah that could work! but maybe just with the days and hours precsions, I don't think its necessary to have minutes and seconds too.

vpetersson, 2017-10-19:
Agreed.  Hours and days feels sufficient.

daniep01, 2017-10-21:
I could create a PR now which allows you to configure the period in days using a numeric text box. Adding hours will take me a bit longer.

vpetersson, 2017-10-21:
Hi @daniep01 - let's hold it until we have the full thing, but feel free to create a PR and then we you can update it when you have that part ready.

ptrlv, 2017-12-06:
Hi, please consider adding a tickbox option for an indefinite period. ie. unscheduled.

vpetersson, 2017-12-06:
@ptrlv that doesn't work with this model, but if you only have one asset, that would in reality be more or less the same.

netjordanlee, 2018-08-10:
Please add an "Always Active" option. This has been a major sticking point for me, there has been a number of times when the end date hasn't been set properly and suddenly our screens are just showing a blank interface...

alexstojda, 2018-08-14:
@netjordanlee please open a new issue/feature request. Don't hijack an existing one :)

closed 2018-08-14 by alexstojda