Screenly/Anthias #635: Screenly Configuration Tool breaks DNS resolution

jonwitts, 2017-09-19:
Hi there,

I am running Screenly OSE on Raspbian Stretch. I used the network configuration tool https://wizard.screenly.io/ to generate the network.ini file to configure my static IP address and DNS. The resulting file has a number of issues which stopped DNS resolution from working and stopped the screenly-net-manager.service from loading.

My downloaded network.ini file is copied at the end of the message. 

The ntp settings stopped the screenly-net-manager.service from loading as the script cannot find the /etc/ntp.conf file

DNS resolution was broken as the network.ini file has commas separating the DNS server names, whereas we need spaces for the servers to correctly import into /etc/network/interfaces and /etc/resolv.conf 

Deleting the NTP line and changing the commas to spaces in the network.ini file and then reboot resolved the issue for me.

Hope this is useful.

----------------------
[eth0]
ip=10.0.103.75
netmask=255.255.255.0
gateway=10.0.103.1

[generic]
ntp=10.0.96.55
dns=10.0.96.55,10.0.96.57,10.0.96.58

vpetersson, 2017-09-20:
Hi @jonwitts - Yes, this is a known issue. We're working on support for Stretch. A lot of things have changed there, and we're not there yet.

jonwitts, 2017-09-20:
Hi @vpetersson thanks for letting me know. I made the relevant tweaks and my system seems to be working fine. I have not seen any other issues from Stretch yet!

closed 2017-09-20 by jonwitts

vpetersson, 2017-09-25:
Awesome. Yeah i wouldn't expect to see any particular issues with Stretch beyond this, but we will keep an eye on it.