Screenly/Anthias #640: Uploading files via POST request

0gener, 2017-09-26:
Hi everyone,

I wonder whether it's possible to upload files to Screenly using a POST request. Basically, we have screen running on port 8080 and a Node app that is constantly checking a git repository for new files. I already can make POST requests to add websites and make them instantly viewable on Raspberry, but whenever I try to POST a file I always get this error:  "\'NoneType\' object has no attribute \'filename\'".
As anyone done something like this or encountered an error like this one?

Best Regards,
Ivan Sousa.

vpetersson, 2017-09-26:
Yes, this should be possible. If you upgrade to the latest development release, you should even have API documentation available at /api/docs

0gener, 2017-09-26:
Thanks for your response @vpetersson . Meanwhile, I managed to find out what was giving me that filename error.
For future reference, when you do the POST it is absolutely necessary to pass the header:
`Content-Disposition: 'attachment; filename="name_of_the_file"'`

vpetersson, 2017-09-26:
Interesting find. We should probably change that as I don't think that makes a lot of sense. Ping @antonmolodykh.

closed 2017-09-26 by 0gener