Screenly/Anthias #642: Install pip dependencies Error

Minimerl, 2017-09-28:
Whenever I try to upgrade my Screenly it fails at "TASK [screenly : Install pip dependencies] and gives me this error:

`TASK [screenly : Install pip dependencies] *************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "/usr/local/bin/pip install -r /home/pi/screenly/requirements.txt", "failed": true, "msg": "stdout: Requirement already satisfied: Mako==0.7.3 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 1))\nCollecting MarkupSafe==0.15 (from -r /home/pi/screenly/requirements.txt (line 2))\nRequirement already satisfied: ansible==2.1.0.0 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: flask==0.12.2 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 4))\nRequirement already satisfied: flask_restful==0.3.6 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 5))\nRequirement already satisfied: flask-restful-swagger-2==0.35 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 6))\nRequirement already satisfied: flask-swagger-ui==3.0.12 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 7))\nRequirement already satisfied: flask-cors==3.0.3 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 8))\nRequirement already satisfied: cffi==1.7.0 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 9))\nRequirement already satisfied: certifi==2015.04.28 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 10))\nRequirement already satisfied: configparser==3.5.0 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 11))\nRequirement already satisfied: gunicorn==19.4.5 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 12))\nRequirement already satisfied: hurry.filesize==0.9 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 13))\nRequirement already satisfied: netifaces==0.10.4 in /usr/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 14))\nRequirement already satisfied: pyasn1==0.1.8 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 15))\nRequirement already satisfied: python-dateutil==2.4.2 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 16))\nRequirement already satisfied: pytz==2012d in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 17))\nRequirement already satisfied: requests[security]==2.10.0 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 18))\nRequirement already satisfied: sh==1.08 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 19))\nRequirement already satisfied: six==1.8.0 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 20))\nRequirement already satisfied: uptime==2.0.2 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 21))\nRequirement already satisfied: werkzeug==0.10.4 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 22))\nRequirement already satisfied: wsgiref==0.1.2 in /usr/lib/python2.7 (from -r /home/pi/screenly/requirements.txt (line 23))\nRequirement already satisfied: futures==3.0.5 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 24))\nRequirement already satisfied: youtube_dl>=2017.7.2 in /usr/local/lib/python2.7/dist-packages (from -r /home/pi/screenly/requirements.txt (line 25))\nCollecting pyzmq==16.0.2 (from -r /home/pi/screenly/requirements.txt (line 26))\n  Using cached pyzmq-16.0.2.tar.gz\nCollecting gevent==1.2.2 (from -r /home/pi/screenly/requirements.txt (line 27))\n  Using cached gevent-1.2.2.tar.gz\nCollecting gevent-websocket==0.10.1 (from -r /home/pi/screenly/requirements.txt (line 28))\nRequirement already satisfied: pycrypto>=2.6 in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: setuptools in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: paramiko in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: PyYAML in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: click>=2.0 in /usr/local/lib/python2.7/dist-packages (from flask==0.12.2->-r /home/pi/screenly/requirements.txt (line 4))\nRequirement already satisfied: itsdangerous>=0.21 in /usr/local/lib/python2.7/dist-packages (from flask==0.12.2->-r /home/pi/screenly/requirements.txt (line 4))\nRequirement already satisfied: aniso8601>=0.82 in /usr/local/lib/python2.7/dist-packages (from flask_restful==0.3.6->-r /home/pi/screenly/requirements.txt (line 5))\nRequirement already satisfied: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi==1.7.0->-r /home/pi/screenly/requirements.txt (line 9))\nRequirement already satisfied: pyOpenSSL>=0.13; extra == \"security\" in /usr/local/lib/python2.7/dist-packages (from requests[security]==2.10.0->-r /home/pi/screenly/requirements.txt (line 18))\nRequirement already satisfied: ndg-httpsclient; extra == \"security\" in /usr/local/lib/python2.7/dist-packages (from requests[security]==2.10.0->-r /home/pi/screenly/requirements.txt (line 18))\nCollecting greenlet>=0.4.10 (from gevent==1.2.2->-r /home/pi/screenly/requirements.txt (line 27))\n  Using cached greenlet-0.4.12.tar.gz\nRequirement already satisfied: cryptography>=1.1 in /usr/local/lib/python2.7/dist-packages (from paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: bcrypt>=3.1.3 in /usr/local/lib/python2.7/dist-packages (from paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: pynacl>=1.0.1 in /usr/local/lib/python2.7/dist-packages (from paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: idna>=2.1 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.1->paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: asn1crypto>=0.21.0 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.1->paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: ipaddress in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.1->paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nRequirement already satisfied: enum34 in /usr/local/lib/python2.7/dist-packages (from cryptography>=1.1->paramiko->ansible==2.1.0.0->-r /home/pi/screenly/requirements.txt (line 3))\nBuilding wheels for collected packages: pyzmq, gevent, greenlet\n  Running setup.py bdist_wheel for pyzmq: started\n  Running setup.py bdist_wheel for pyzmq: finished with status 'error'\n  Complete output from command /usr/bin/python -u -c \"import setuptools, tokenize;__file__='/tmp/pip-build-5FCYJY/pyzmq/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\\r\\n', '\\n');f.close();exec(compile(code, __file__, 'exec'))\" bdist_wheel -d /tmp/tmpJ1LzHkpip-wheel- --python-tag cp27:\n  running bdist_wheel\n  running build\n  running build_py\n  creating build\n  creating build/lib.linux-armv7l-2.7\n  creating build/lib.linux-armv7l-2.7/zmq\n  copying zmq/error.py -> build/lib.linux-armv7l-2.7/zmq\n  copying zmq/decorators.py -> build/lib.linux-armv7l-2.7/zmq\n  copying zmq/__init__.py -> build/lib.linux-armv7l-2.7/zmq\n  creating build/lib.linux-armv7l-2.7/zmq/devices\n  copying zmq/devices/basedevice.py -> build/lib.linux-armv7l-2.7/zmq/devices\n  copying zmq/devices/proxydevice.py -> build/lib.linux-armv7l- [... cut ...]
NO MORE HOSTS LEFT *************************************************************

RUNNING HANDLER [restart-x-server] *********************************************
        to retry, use: --limit @site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=34   changed=7    unreachable=0    failed=1

`

Already tried several solutions for similar problems but couldn't fix it.

I actually just want the login with pw, I would be ok with an older version, but older solutions for this just caused crashes.

vpetersson, 2017-09-28:
@Minimerl What Raspbian version are you on?

Minimerl, 2017-09-28:
It's a Raspberry 3 B 
And I installed Screenly OSE via noobs (if that matters)

anayrat, 2017-09-28:
It seems duplicate of #627

antonmolodykh, 2017-10-01:
@anayrat @Minimerl  I was able to reproduce the problem. for some reason it does not always appear (I reinstalled and tried to reproduce it several times). I'll let you know when I understand the reasons and make the fix

antonmolodykh, 2017-10-05:
@anayrat @Minimerl we made a Fix in the master branch. please try to upgrade now

vpetersson, 2017-10-09:
Closing out as it has been fixed. Please comment if it did not solve the issue.

closed 2017-10-09 by vpetersson