Screenly/Anthias #645: Allow for Mass Enable/Disable of Assets

aolszowka, 2017-10-09:
It would be nice if there was a better interface for mass updating the enabled/disabled state of the assets.

This can be worked around by poking the SQLite database in the background.

My use case was as follows:
I am trying to setup a slide show to play disconnected with several hundred files. It would be very painful to go through and enable each one of these assets via the Web Interface.

vpetersson, 2017-10-10:
I don't think it would make sense to do this in the web interface. Instead, take a look at the API documentation. I believe that it should be possible to do this over the API already.

mflage, 2019-07-17:
> I don't think it would make sense to do this in the web interface. Instead, take a look at the API documentation. I believe that it should be possible to do this over the API already.

I don't think it's possible to do this through the API. I found the following: https://www.raspberrypi.org/forums/viewtopic.php?t=12396&start=925 that did something like this, but the /api/v1/assets/<asset-id> doesn't support the POST method. You can therefore only get information of the asset, but not change it in any way. Is it possible to get this working again?

vpetersson, 2019-07-17:
Correct, you cannot do a POST on an asset. That's by design, because there are a number of checks when an asset is added. Hence the DRY approach is to do:

* DELETE on /api/v1.1/assets/{asset_id} 
* POST on /api/v1.1/assets

The API docs can be found [here](http://ose.demo.screenlyapp.com/api/docs/).

mflage, 2019-07-17:
The problem is that the asset is an uploaded mp4. Does this mean I have to keep uploading the movie every time I want to enable the asset? Something is very off here by design :)

vpetersson, 2019-07-17:
Well, i thought we were talking about *updating* the asset, as in changing the content. If we're merely talking about toggling the `is_enabled` status, that's a different matter.

aolszowka, 2019-07-17:
In the original ask it was for toggling is_enabled

vpetersson, 2019-07-17:
@rusko124 Do we have this feature?

rusko124, 2019-07-18:
@aolszowka use `PUT` method for asset updating.
To enable example:
```
{
  "name":"les_derevia_mrachnyj_125841_1920x1080.jpg",
  "mimetype":"image",
  "uri":"/home/pi/screenly_assets/ce1101f57e5f4d62af7efbf37f3ad1b0",  
  "is_active":0,
  "start_date":"2019-07-18T06:54:00+00:00",
  "end_date":"2019-08-17T06:54:00+00:00",
  "duration":"10",
  "is_enabled":1,
  "is_processing":0,
  "nocache":0,
  "play_order":0,
  "skip_asset_check":0,
  "asset_id":"ce1101f57e5f4d62af7efbf37f3ad1b0"
}
```

To disable example:
```
{
  "name":"les_derevia_mrachnyj_125841_1920x1080.jpg",
  "mimetype":"image",
  "uri":"/home/pi/screenly_assets/ce1101f57e5f4d62af7efbf37f3ad1b0",  
  "is_active":1,
  "start_date":"2019-07-18T06:54:00+00:00",
  "end_date":"2019-08-17T06:54:00+00:00",
  "duration":"10",
  "is_enabled":0,
  "is_processing":0,
  "nocache":0,
  "play_order":0,
  "skip_asset_check":0,
  "asset_id":"ce1101f57e5f4d62af7efbf37f3ad1b0"
}
```

mflage, 2019-07-18:
Perfect, I can confirm that this works as intended. Finally I can remote control my screenly properly. This should however be documented a bit better in the API docs, as the documentaton says: "Adds an asset". It should be changed to "Adds or modifies an asset" or similar.

vpetersson, 2019-07-18:
There's definitely room for improvement here. We should have a simple PATCH on the asset to allow you to *just* set the flag. @rusko124 can you create a ticket for that?

vpetersson, 2019-07-30:
@mflage Fix has now been applied to the master/developer branch.

closed 2019-07-30 by vpetersson