Screenly/Anthias #660: Assets

rsenio, 2017-10-18:
Downloaded the disk image from https://github.com/screenly/screenly-ose/releases. 

2017-10-16-Screenly_OSE_4GB.zip

Set it up, added two assets and watched them play. Was prompted on the next reboot stating there was an update. Performed the update as the Pi user 
bash <(curl -sL https://www.screenly.io/install-ose.sh)

Assets play, but visiting the web UI shows The operation failed. Please reload the page and try again and no assets listed even though they are playing. If I click System Info I see the assets cycling through, but at the top (only in system info) I see "update available"

vpetersson, 2017-10-18:
That was solved today. Try upgrading to the latest development version.

vpetersson, 2017-10-18:
Re-reading the issue, i'm not sure what the operation fail is. Could you provide a screenshot?

closed 2017-10-18 by vpetersson

rsenio, 2017-10-18:
Some sort of strange browser cache issue, but only for the asset page. It seems to be working for now