Screenly/Anthias #662: [RFE] Control the Assets by hand

3PO, 2017-10-22:
Hi,

is it possible to implant a function that enables to control the Assets by hand?

Normally the Assets will shown automaticly, but i want to call the Assets e.g. by a keypress, or a ir-remote control (-> lirc).

vpetersson, 2017-10-22:
You can't disable the transition, but you could do something like this:
 * Set the duration to something very high (like an hour)
 * Use the [control API call](http://ose.demo.screenlyapp.com/api/docs/) to navigate

You'd have to write some kind of event handler for LIRC or keys for this tho.

3PO, 2017-10-22:
Thanks for the response.

When i do,

`curl -s http://localhost:8080/api/v1/assets/control/previous`
or
`curl -s http://localhost:8080/api/v1/assets/control/next`

it works fine, but i'am not able to select a asset directly.

I have 4 assets,
```
raspberrypi ~ # curl -s http://localhost:8080/api/v1/assets | python -m json.tool |grep "asset_id" |cut -d \" -f4
2a780f8b3abb4d458c2730ec8e1c2136
1ebd3f91d3944fa798ea38438d3b29a6
4e87566490964aed8122cb34fb6682b6
0693aefa873e40cd8c75c4be57b360f2
```
but if i want to switch to "Asset 2", this

`curl -s http://localhost:8080/api/v1/assets/1ebd3f91d3944fa798ea38438d3b29a6`

doesn't work. :(

What do i wrong?

vpetersson, 2017-10-22:
Right, so there's currently no API call for navigating to a particular asset. It is however a good idea to add. I'm not sure if we have the bandwidth to squeeze that into the next sprint.

3PO, 2017-10-22:
Nice, but maybe you have the "bandwidth" to extend the API to start/stop the slideshow? ;)

vpetersson, 2017-10-23:
No, that would unfortunately require a equally (or possibly even more) work.

vpetersson, 2017-12-03:
I'm thinking that it should be possible to navigate to *any* asset (enabled or disabled etc) and once the asset has been displayed it would return to the regular playlist.

closed 2017-12-14 by vpetersson