Screenly/Anthias #666: Any issue running Screenly all day long 24/7?

john-bonachon, 2017-10-24:
I wonder if you had some experience running Screenly all day long 24/7.
Is there any problem if TV is turned off at night while Screenly is still running? Is it the same as if the TV is turned on all day long? Should I make a crontab to reboot the Raspberry 2 every day?

vpetersson, 2017-10-25:
@john-bonachon - Screenly is designed to run 24/7. If you turn the monitor off at night (e.g. using a timer) that's fine. The only issue you could run into is if the Raspberry Pi reboots while the TV is turned off. In that case, the resolution is likely to go wrong (and it would require another reboot while the monitor is on). 

There's no need for a cronjob that reboots the device.

closed 2017-10-25 by vpetersson

john-bonachon, 2017-10-25:
Got it..
Do you think that's solved by uncommenting "force_hdmi_hotplug=1" on raspberry's config.txt? I think it was talked before on another issue here.

Maybe an issue would be that the SD card performance gets worst faster..not sure if leaving screenly logging turned on or off

vpetersson, 2017-10-25:
> Do you think that's solved by uncommenting "force_hdmi_hotplug=1" on raspberry's config.txt? I think it was talked before on another issue here.

Sometimes it does, but not always. It's worth a shot. 

> Maybe an issue would be that the SD card performance gets worst faster..not sure if leaving screenly logging turned on or off

SD card lifespan varies a lot. We've done a lot of improvement in Screenly Pro in regards to reducing writes to the SD card. Some of these made it to OSE. Generally speaking, expect that the SD card *will break* at some point. If that's after 3, 6, 12 or 24 months is the real question. There are a large number of variables that factors into this.

We've found that Sandisk and Samsung (class 10) sd cards tend to have a reasonably good lifespan.