Screenly/Anthias #667: Things become stuck at "INACTIVE ASSETS".

zsamenfeld-encon, 2017-10-30:
That's right, I can click the wonderful on and off button all I'd like but it doesn't seem to do anything. ever.
I'm not totally sure what caused this because earlier today the screen still had assets showing. Now it's just the screenly logo faintly in the background. A restart of the pi seems to fix this.

[image]

vpetersson, 2017-10-31:
My gut feeling is that there is some issue with the clock. Is this device offline?

zsamenfeld-encon, 2017-10-31:
Time reads, Tue Aug 8 16:00:15 BST 2017. Clearly wrong.. I wonder how that's happened.

vpetersson, 2017-10-31:
The Raspberry Pi does not have a built-in real time clock (RTC). Hence, if it is offline and unable to communicate with an NTP server, it will not know what time it is. See [this FAQ entry](https://support.screenly.io/hc/en-us/articles/115002570873-Can-I-use-Screenly-OSE-in-offline-mode-) for more details.

closed 2017-10-31 by vpetersson

zsamenfeld-encon, 2017-11-07:
Alright, I've installed a RTC on the raspberry pi, configured it.. etc. Still having inactive asset issue randomly.

vpetersson, 2017-11-07:
Did you verify that the time was properly set? It's likely you need to do this by hand (or via NTP) the first time.

zsamenfeld-encon, 2017-11-07:
Yes, I've made sure the time is set properly.

vpetersson, 2017-11-08:
The next thing then is to look at the logs. Can you post the output of the following commands:

 * `sudo journalctl`
 * `date`
 * `dmesg`

zsamenfeld-encon, 2017-11-08:
https://gist.github.com/zsamenfeld-encon/a29e3b3252aec9be337fb97c47f68e0a

vpetersson, 2017-11-09:
Nothing obvious in the logs unfortunately. When was the last time you upgraded?

Also ping @antonmolodykh.

zsamenfeld-encon, 2017-11-09:
At the time of uploading the logs I wasn't having any issues, I'll report back with logs again when it happens again.
Also we upgrade every Sunday night. So the last update was only a few days ago.