Screenly/Anthias #669: Jenkins BlueOcean pages not showing

michburkhalter, 2017-11-03:
I try to have the status of my Jenkins Blueocean pipeline builds visualized with Screenly OSE.

When I configure a URL like http://xxx.xxx.x.x:8080/blue/organizations/jenkins/pipelines Screenly OSE will try to load this page forever but never show anything. The screen is white and the progress bar on top is getting slower and slower.
If I enter the same address in my browser, the page is displayed correctly.

This is probably related to how jenkins blue delivers its content.

vpetersson, 2017-11-03:
Yeah this is most likely related to the fact that Jenkins is using something that UZBL can't handle unfortunately. Unfortunately it's not an easy thing to solve in Screenly OSE. See #527.

michburkhalter, 2017-11-03:
Thanks for the answer, @vpetersson. I would have loved to use screenly to visualize our build status..

vpetersson, 2017-11-03:
Yeah, that's a great use case for Screenly. We'll have to look if we can make some headway on #527 some time in the near future.

vpetersson, 2017-11-21:
Merging this one into #527.

closed 2017-11-21 by vpetersson

vpetersson, 2018-06-22:
@michburkhalter Could you try with the experimental branch that we just shipped?

michburkhalter, 2018-07-16:
@vpetersson Which expiremental branch do you mean?

antonmolodykh, 2018-07-16:
@michburkhalter @vpetersson talks about the experimental version of Screenly OSE that we recently published. It includes the new browser with hardware acceleration support (as a replacement for old uzbl), and migration of all services to the docker. To try it follow the steps under 'Manual upgrade' [here](https://www.screenly.io/ose/) and answer yes to the question about experimental version.

michburkhalter, 2018-07-16:
@antonmolodykh yeah I just saw that I can choose Experimental when I run the installation script. I'll try it when I have 5 minutes to spare.

michburkhalter, 2018-07-17:
@antonmolodykh @vpetersson I ran the installation script and selected Experimental. If I cd into home/pi/screenly and hit "git status" it tells me "_On branch experimental_" and "_up-to-date with 'origin/experimental_'".
Unfortunately the Jenkins Blueocean Pipeline builds page is still not fully loaded. No change to the previous version.

vpetersson, 2018-07-17:
@michburkhalter You don't happen to have a public Blueocean instance that we could test against?

michburkhalter, 2018-07-17:
@vpetersson Nope, sorry. Our Blueocean instance is not public.