Screenly/Anthias #674: Fix final Resin issues

vpetersson, 2017-11-12:
I was playing with the Resin addition today and found a few issues

* Update documentation for how to use remote URL. There also appears to be an issue since we use port 8080
* It appears as the [tests now fail](https://travis-ci.org/Screenly/screenly-ose/builds/301023439?utm_source=email&utm_medium=notification)
* There is no mention of `DISABLE_UPDATE_CHECK` in the documentation

imrehg, 2017-11-13:
The remote URL currently can be used:

* Screenly server running on port 8080, and then use "http://<uuid>.resindevice.io:8080", manually setting HTTP and the port
* or internally redirecting from port 80->8080 in the resin application, and manually set HTTP ("http://<uuid>.resindevice.io").

For the default Public URL to work without manual setting, would need to enable to use wss:// for websockets (instead of unencrypted ws://) and adding the internal redirect.

vpetersson, 2017-11-13:
@imrehg - Is it  possible to do port mapping in Docker? We're running on 8080 because we drop permission to a runtime user rather than running as root.

imrehg, 2017-11-13:
@vpetersson Yes, understood the permission drop. The port mapping was part of the original example I've sent, with iptables setting up the redirection https://github.com/resin-io-playground/resin-screenly-ose/blob/master/startup.sh#L45-L59 This is I guess not the only way, but it was working well enough when I've tested it.

vpetersson, 2017-11-13:
@imrehg We support HTTP/HTTPS internally (by using Nginx as a reverse proxy). If we enable this on a self-signed cert, would that work?

vpetersson, 2017-11-14:
@imrehg -  ignore the issue above with Nginx. I take it that your proxy reverse proxy from 443 to 80 on the device only. As such, there is no need to enable SSL on the device.

@antonmolodykh - in regards to @imrehg comment above about `ws://` vs `wss://`, we  need to fix this as it affects `enable_ssl.sh` as well. How about something like this:

 * When the user runs `enable_ssl.sh`, we add a new key to `screenly.conf`:
```
admin_ssl = True
```

 * If this key is present, we use `wss://` instead of `ws://` for the websocket
 * We can then piggyback on this same logic for Resin by simply adding this key

vpetersson, 2017-11-14:
Also, as discussed with @antonmolodykh - we need a new Nginx config that only reverse proxies 80 -> 8080 on the device.

vpetersson, 2017-11-14:
@imrehg - one question for you - since we're using systemd now to manage processes, we don't get any logging in the Resin interface. Is there an existing solution for this that you guys have been using?

vpetersson, 2017-11-14:
Outstanding issues:
 
 * Conditional `ws://`/`wss://` switching
 * Logging to stdout (?)

imrehg, 2017-11-14:
@vpetersson The logging was addressed in the original setup I've sent you, the start script can finish with `journalctl -f -a` which would output the systemd logs into the dashboard. Or you can modify it to just output some units only (`journalctl -f -a -u <unitname>`...). Just not put anything after after that in the script, as that command will just keep on going.

https://github.com/resin-io-playground/resin-screenly-ose/blob/master/startup.sh#L61-L68

vpetersson, 2017-11-14:
Awesome. Thanks.

@antonmolodykh - let's add this so that the logs for all Screenly processes are appended to the Resin logs.

vpetersson, 2017-11-17:
@antonmolodykh Did you confirm that everything works on WiFi? If so we can close it out.

antonmolodykh, 2017-11-17:
Yes it works. right now I use Screenly with Resin without ethernet cable

vpetersson, 2017-11-17:
👍

closed 2017-11-17 by vpetersson