Screenly/Anthias #682: Issue deploying screenly with Resin.io

mckaycr, 2017-11-18:
I'm getting the following error in the UI.
`Server Error: attempt to write a readonly database`
I attempted to deploy to an RPI Zero W and an RPI3.

I'm new to resin, like new as in my first time today so its possible i'm doing something wrong.  I basically cloned the screenly master repo to my local computer, then build and uploaded to resin.  Resin deployed it back to my devices

Here is my log from the RPI3 attempt
[rasplex_little-forest-17.11.17_22_04_45_(-0500).txt](https://github.com/Screenly/screenly-ose/files/1484320/rasplex_little-forest-17.11.17_22_04_45_.-0500.txt)

vpetersson, 2017-11-18:
Ping @antonmolodykh

vpetersson, 2017-11-20:
@mckaycr try now. We just landed ##683.

vpetersson, 2017-11-20:
Closing out this one. Please comment if you're still having issues, @mckaycr.

closed 2017-11-20 by vpetersson

mckaycr, 2017-11-20:
Thanks, everything looks good now.