Screenly/Anthias #686: Add resin-wifi-connect

mckaycr, 2017-11-21:
Wondering if it might add any value to add the resin-wifi-connect?

vpetersson, 2017-11-21:
Interesting idea. I don't think it would be compatible with regular Raspbian, so we'd have to make it unique to the resin.io Dockerfile. Perhaps @imrehg can chime in.

mckaycr, 2017-11-21:
Their documentation says it has also been successfully tested with the on-board WiFi on a Raspberry Pi 3, not sure if that helps.

vpetersson, 2017-11-21:
Yeah the problem isn't that. The problem is to integrate it into the stack.

mckaycr, 2017-11-21:
Makes sense.  Its not a must have, its a nice to have, so if it doesn't make it in it's no big deal.

vpetersson, 2017-11-21:
Happy to accept a PR if you want to hack on it @mckaycr.

imrehg, 2017-11-22:
@vpetersson it should not require a resin base image to work as it stands now, so should be possible to integrate. One thing, though, not sure if the access point interface's port is configurable or not, since here both port 80 and 8080 are in use. @majorz might now a bit more.

majorz, 2017-11-22:
The port is not configurable yet, but I thihnk we can add such option.

Latest version of WiFi Connect runs on Raspbian Stretch, but requires switching the network manager from dhcpcd to NetworkManager, which maybe will not suit your use case. We have installation script that does that: https://github.com/resin-io/resin-wifi-connect/blob/master/scripts/raspbian-install.sh
Documentation about this will land soon together with other improvements.

vpetersson, 2017-11-25:
Thanks @imrehg / @majorz. Good pointers.

> [...]  but requires switching the network manager from dhcpcd to NetworkManager, which maybe will not suit your use case.

I don't see that as a problem. Should work fine.

> Documentation about this will land soon together with other improvements.

Excellent. Basically what we need is to figure out the logic. Do you guys have any thoughts on this? Basically I think it would make sense to only display/activate this hotspot when there is no connection/DHCP found through Ethernet.

majorz, 2017-12-01:
Hey, sorry for the delayed response. I missed somehow the notification earlier this week and found it now by checking my backlogs :/

I will be working on customizing the port number today and which will be included in the version we are about to release next week.

Yes, it makes sense to run WiFi Connect only when similar condition exists. I will think a bit more about that and follow up soon!

vpetersson, 2017-12-03:
I had a chat with @curcuz about this, and [this is the way](https://github.com/resin-io-projects/boombeastic/blob/f91b844f6b492035080cca945d011c321d53a9d2/app/start.sh#L24-L33) he implemented the logic in Boombeastic.

I think we can improve upon this logic a bit by doing something like if there is an active link on either wlan0 or eth0 (i.e. we received an IP), then we skip the WiFi portal. We'd need some additional things around that to make sure we sleep etc such that we have enough time to connect. We might also want to have a password set or similar to avoid random takeovers.

majorz, 2017-12-11:
The customizable port number feature is not as straightforward as I expected. Starting the WiFi Connect webserver on a different port is easy, however when you connect to the WiFi Connect network with your phone, the phone OS triggers a captive portal check request to determine whether the network is behind a captive portal. Currently we use dnsmasq to resolve all host names to the captive portal IP address, but the request still targets port 80. Some additional iptables rules will be needed for the port number redirect and I will have to bring it up to our team for further discussion. I will post update when I have more information on this.

One workaround for the port number issue is to start WiFi Connect before starting the other application that listens on port 80, or to stop temporary the other application when WiFi Connect should be run. After WiFi credentials are entered and connection is established, WiFi Connect will exit and the other application can be started.

A different use case is where for example a button is used to start WiFi Connect when new WiFi credentials have to be entered while the main application is already running. This is the type of use case where the customizable port number functionality is actually needed, since both application have to run at the same time.

Can you please let me know which of the above two scenarios the Screenly use case falls into?

I am collecting also a set of examples for conditions for running WiFi Connect (which will be included in the next coming soon release) and I will explore the default gateway condition you suggest as well. Again I will make sure to follow up on this here.

majorz, 2017-12-11:
I assembled a short list of checks that can be used in a way similar to how Boombeastic does the google ping:

1. Is there Internet connectivity?
`nmcli -t g | grep full`

2. Is there a default gateway?
`ip route | grep default`

3. Is there an active connection?
`nmcli -t c show --active | grep :`

4. Is there an active WiFi connection?
`nmcli -t c show --active | grep 802-11-wireless`

vpetersson, 2017-12-14:
Thanks for the pointers, @majorz. I spent some time thinking about this, and I think this workflow would be sensible for a first-boot:

* We check if there is any active internet connection. I think looking for the default gateway would be a sensible approach for that.
* If there is no connection, we'd do the following:
  * Stop Nginx (as it binds on `:80`).
  * Start the access point.
  * Display a custom screen (e.g. `/hotspot`) with instructions how to connect to the device (SSID etc).
  * Wait for the user to configure the network using the hotspot.
* Shut down resin-wifi-connect and bring Nginx back up and proceed with the regular boot order.

@majorz / @antonmolodykh what do you guys think?

closed 2017-12-29 by vpetersson