Screenly/Anthias #687: Unable to configure Wifi on Screenly-ose

ChantalD, 2017-11-24:
I'm trying to evaluate what will be our best need (between OSE and PRO version). I'm using same Raspberry Pi 3b and have 2 SD CARDS. The Wifi network is working when deployed with network.ini on PRO version but it isn't when deployed on OSE version. I'm using same Pi, same Wifi Network and same Network.ini file. Is there a tip you can give me to have Wifi working on OSE Version?

Content of network.ini file
[wlan0]
ssid=SSIDWifi
passphrase=NotTheRealPwd
mode=dhcp

Image use for Screenly OSE -> 2017-10-16-Screenly_OSE_4GB

Thank you

vpetersson, 2017-12-14:
Hi @ChantalD,

Sorry for the late response. I missed the notification for this.

In short, we are moving away from using this tool in Screenly OSE (#691). We're also making some changes to the way we deal with network configuration in Screenly Pro going forward.

To answer your question however, I would recommend that you simply disable the Screenly Network Manager (`systemctl disable screenly-net-manager.service`) and then follow [these](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md) instructions.

closed 2017-12-14 by vpetersson

frankleggett, 2018-02-07:
Hi,

I have a raspberry pi running Screenly OSE and I want to add wifi capability. I have changed the wpa_supplicant.conf file to my SSID and Password but I can't turn off net-manager as it is asking for authorisation and then a password which I haven't set or don't know.

Please can someone assist with this issue.

Many thanks
Frank
[image]

vpetersson, 2018-02-07:
@frankleggett As of the [latest release](https://www.screenly.io/blog/2018/01/19/screenly-ose-0.13-is-available/), Screenly Net Manager has been decommissioned.

As far as connecting to WiFi, I'd recommend that you use the new method. Alternatively, you can also use [Network Manager](https://help.ubuntu.com/community/NetworkManager?action=show&redirect=WifiDocs%2FNetworkManager), which admittedly, is a bit clunky to use by hand.

frankleggett, 2018-02-08:
I found reloaded screenly with the latest development version but when I disconnect the lan cable at startup it just goes to a white screen.

I did find this link to wifi-connect-resin 
https://github.com/resin-io/resin-wifi-connect

and it has the following instructions

"Run the following in your terminal, then follow the onscreen instructions:

curl https://github.com/resin-io/resin-wifi-connect/raw/master/scripts/raspbian-install.sh -sSfL | bash"

obviously this doesn't work if you copy it literately so could someone please convert it into the correct syntax

Many thanks
Frank

vpetersson, 2018-02-09:
@frankleggett interesting - What did you use as the base image? We'd be keen to reproduce this ourselves. (ping @antonmolodykh).

frankleggett, 2018-02-09:
This is a noobs image of raspbian then I did an install of screenly over terminal. I haven't tried to create an image of screenly on a sd card. I apologise that my knowledge is very basic at present.

I was able to get my first pi to work by installing wifi-connect but I had to try several different versions of the instruction line. I then installed screenly and selected the development version and this worked and the wifi-connect option functioned.

I tried again with another screenly installed pi and this is where I came across the white screen. I have since down a full update and this now flashed up with a screenly logo then just goes to a black screen. It is still contactable via Putty if I connect a LAN cable.

vpetersson, 2018-02-09:
@frankleggett If you're using NOOBS, you'd do yourself a favor by simply using the Screenly installer available through NOOBS. That already includes the latest updates.

frankleggett, 2018-02-09:
I tried that but it doesn't enable VNC, SSH or remove underscan. I have found it easier to install raspbian enable ssh, vnc and underscn and then install screenly via Putty after updating and upgrading raspbian. I now have 2 working pi with wifi capability and the 3rd is installing now.