Screenly/Anthias #688: Add auto-resize of disk on first boot

vpetersson, 2017-11-25:
There is a functionality in Raspbian to resize the disk on first boot. We should try to invoke this during the build process.

Let's read the code startup code for Rasiban and see what is triggering this behavior.

vpetersson, 2018-01-19:
I can confirm that this does not work. You can reproduce it with the latest disk image.

antonmolodykh, 2018-01-20:
@vpetersson in the new release there is no `init=/usr/lib/raspi-config/init_resize.sh` in the `/boot/cmdline.txt` file. I think that the cause of the problem in this.

vpetersson, 2018-01-20:
There is. I'm 99.9% confident as I added it before. It will however disappear after first boot. If you mount the disk image, you should be able to see it there.

antonmolodykh, 2018-01-20:
I just downloaded a new image and checked the file. I did NOT run the device, I checked it using computer.

vpetersson, 2018-01-20:
Strange. Ok, I guess we might have to do a hotfix then.

vpetersson, 2018-01-20:
I stand corrected. You're right. I screwed up. WIll do a new image.