Screenly/Anthias #691: Disable Screenly Network Manager

vpetersson, 2017-11-27:
It seems like there has been a fair amount of issues with Screenly Network Manager on Screenly OSE. Let's disable this in favor of #686.

vpetersson, 2017-11-27:
@antonmolodykh Can you provide instructions for how to disable Screenly Network Manager (the `systemctl` commands) for future references?

antonmolodykh, 2017-11-27:
To just turn off the service, just enter `systemctl disable screenly-net-manager.service` command

antonmolodykh, 2017-12-25:
I think that we can not completely abandon this, because as @majorz [said](https://github.com/Screenly/screenly-ose/issues/686#issuecomment-346333810), resin-wifi-connect is now being run on Stretch. so we need to leave support for Jessie.

vpetersson, 2017-12-27:
@antonmolodykh - I think that's fine. We will just revert to the default method in Raspbian for earlier releases.

vpetersson, 2017-12-29:
Solved in #715.

closed 2017-12-29 by vpetersson