Screenly/Anthias #692: Screenly OSE often starts in console mode

zuendi123, 2017-11-28:
I have installed the newest Screenly OSE release from November 25th, 2017.
It happens very often that Screenly OSE starts in console mode (waiting for login credentials).
When I restart the Pi it often starts in playlist mode but sometimes it starts in console mode again.
Is there a fix for the problem? How can I start the playlist via remote (e.g. PuTTY)?

vpetersson, 2017-11-28:
My best guess is that it's either a broken SD card or a corrupt download.

I'd recommend the following steps:

 * Re-download the disk image from Githug
 * Verify the download with the MD5 file
 * Download [SD Formatter](https://www.sdcard.org/downloads/formatter_4/) and do an 'overwrite format' of the SD card
 * Re-flash the SD card using Etcher

coan29001, 2017-11-28:
I experience this issue too. The first time I boot the image everything is working fine.
However, the second time it goes directly to console.

vpetersson, 2017-11-28:
Hmm that's odd. @antonmolodykh are you able to reproduce this?

zuendi123, 2017-11-28:
@vpetersson Can you provide me a command for starting the playlist or for a clean restart of the Pi?
The command "sudo shutdown -r 0" doesn't work right, I have to disconnect and reconnect the power cable every time.

vpetersson, 2017-11-28:
@zuendi123 there are no command for starting the playlist. If you get to that state, something is *severely* wrong with the setup. What do you get if you run `journalctl`?

antonmolodykh, 2017-11-28:
I'll try to reproduce and let you know

antonmolodykh, 2017-11-28:
@zuendi123 For a simple PI restart, you can type `sudo reboot`.
please specify on which device do you have this problem?

zuendi123, 2017-11-28:
@antonmolodykh I'm running Screenly OSE on a Raspberry Pi (1) Model B+ 512MB.
The `journalctl` output is in progress, I'm trying to reproduce the problem again.

vpetersson, 2017-11-28:
@zuendi123 If you're running on such old Raspberry Pi, try reducing the GPU memory split in `raspi-config` to something like `96` or similar to see if that resolves the issue.

vpetersson, 2017-11-28:
Note-to-self: Use proper device-specific GPU configs (https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md)

zuendi123, 2017-11-28:
Here is the output from `journalctl`:
[journal.txt](https://github.com/Screenly/screenly-ose/files/1509654/journal.txt)

@antonmolodykh `sudo reboot` doesn't work for me. The playlist stops and I get a black screen but nothing happens.

@vpetersson I'll try that later.

vpetersson, 2017-11-28:
The GPU split is my best lead right now, but could you also try running `free -m` on the device?

vpetersson, 2017-11-28:
@coan29001 What Raspberry Pi hardware are you running on?

zuendi123, 2017-11-28:
I changed the split memory to `96`. I'll keep an eye on this and report potential changes later.

@vpetersson `free -m` shows following output:
```
pi@raspberrypi:~ $ sudo free -m
              total        used        free      shared  buff/cache   available
Mem:            402         111         125          12         166         231
Swap:             0           0           0
```

vpetersson, 2017-11-28:
Cool. Yeah if you load a website or something along those lines, that would be sufficient to push it over in to the land of OOM. ;)

zuendi123, 2017-11-28:
Ok, so would you suggest to minimize the split memory even more to `64` or something?

vpetersson, 2017-11-28:
is that after you changed it to 96 and rebooted? If so, let's see what happens. If it keeps happening, then yes, drop it down further.

zuendi123, 2017-11-28:
Yes it is.
Ok, so I'll keep this in mind like I said.
Thank you for now ;)

zuendi123, 2017-11-29:
Newest details:
I started a system update with `bash <(curl -sL https://www.screenly.io/install-ose.sh)`.
On the first run, it showed some errors (e.g. `Not a directory: /usr/share/info`, see #479).
After creating the directory `info` in `/usr/share` and rerunning the update script, there were no errors anymore. But after a reboot the console screen came up again. Had to do a reboot with `sudo reboot`, which worked for now.

vpetersson, 2017-11-30:
hm definitely sounds like a corruption issue. Did you verify the download with the MD5 file? Also, did you try a different SD card?

zuendi123, 2017-12-01:
I re-flashed the SD card because of WLAN not functioning. Before I formatted it with SD Formatter like you posted in your first answer.
The download is verified with the MD5 file. I did not try an other SD card yet because I don't have one.

vpetersson, 2017-12-04:
What do you get if you run `ifconfig -a` and `iwconfig` on the device?

vpetersson, 2017-12-11:
Closing out this one since we're unable to reproduce it.

closed 2017-12-11 by vpetersson