Screenly/Anthias #703: Screen Displays Black

jpburnett, 2017-12-07:
The Pi I have running screenly just displays black. I can go to the terminal and it will display thing just fine. I reboot the pi and it will display the screenly logo but shortly after will just display black. I've checked on the web interface to verify that there are assets to display.

vpetersson, 2017-12-11:
What Raspberry Pi hardware are you using?

antonmolodykh, 2017-12-11:
hmm, I confirm the problem. I just noticed that after a while the screen just turned black. and when I inserted the usb adapter (keyboard) into PRi, the assets reappeared on the screen

jpburnett, 2017-12-11:
I have a Raspberry Pi model 3 hooked up to NEC 42" TV with a wired internet connection. I have this similar set up with 4 other raspberry Pis on 2 other brands of TVs. One station manages to have no issues but the other 3 do. I have tried writing a cron job to reboot the Pis every morning and that doesn't work. I tried re-imaging the Pis and starting from scratch several times but there is one that always manages to go black.

vpetersson, 2017-12-14:
@antonmolodykh Do you have a solution for this?

@jpburnett Did you try a different SD card? We shouldn't rule that out yet.

antonmolodykh, 2017-12-14:
@vpetersson no, I did not understand it yet. @jpburnett I'll let you know here when I find something

jpburnett, 2017-12-18:
@vpetersson I tried a new SD card and working with the screen it blacks out still. When I rebooted the Pi it gets stuck on the boot up screen. The last line it prints is
2.094898]---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x000000000b

vpetersson, 2017-12-18:
Hmm. Did you verify the MD5 file of the download? Something seems corrupt by the sound of it. Also, did you use Etcher to flash it out? It provide automatic verification of the disk image.

vpetersson, 2017-12-29:
@jpburnett closing out issue due to lack of response.

jpburnett, 2017-12-29:
@vpetersson was @antonmolodykh still not looking in to the issue? I did what you said and even with screenly reinstalled the issue still persists.

vpetersson, 2017-12-29:
@jpburnett See above. My best guess is that this is related to a corrupt download.

jpburnett, 2018-01-09:
@vpetersson I imaged the Pis using SD card formatter and downloaded the image from 11-25. I imaged it using etcher and I am still seeing the issue of a screen going black. I can access the pi from ssh and access the terminal using a keyboard pressing ctrl+alt+f2, so the Pi is working. I reboot the system and the assets will display but the screen will always go black some time later.

vpetersson, 2018-01-09:
Can you do `journalcfl > foobar.txt` and attach `foobar.txt` once it is showing a black screen?

jpburnett, 2018-01-10:
[sysReport.txt](https://github.com/Screenly/screenly-ose/files/1620309/sysReport.txt)
@vpetersson @antonmolodykh Attached is the report. I wasn't able to access the online interface (which makes sense since the log said it's stuck in the failed state).

vpetersson, 2018-01-12:
Very strange error. Something causes the viewer to crash.

```
Jan 10 03:25:52 raspberrypi systemd[1]: Stopped Screenly Web UI.
Jan 10 03:25:52 raspberrypi systemd[1]: Starting Screenly Web UI...
Jan 10 03:25:52 raspberrypi systemd[1]: screenly-web.service: Control process exited, code=killed status=11
Jan 10 03:25:52 raspberrypi systemd[1]: Failed to start Screenly Web UI.
Jan 10 03:25:52 raspberrypi systemd[1]: screenly-web.service: Unit entered failed state.
Jan 10 03:25:52 raspberrypi systemd[1]: screenly-web.service: Failed with result 'signal'.
Jan 10 03:25:57 raspberrypi systemd[1]: screenly-web.service: Service hold-off time over, scheduling restart.
```

Any chance you could send over the entire dump of the config?

```
$ tar cvfz srly.tgz .screenly screenly screenly_assets
```

If it includes sensitive data, we could send it through an alternative channel

jpburnett, 2018-01-15:
I tried uploading the srly.tgz but it looks like the format isn't supported with GitHub. How else would you like me to send you it?

vpetersson, 2018-01-16:
Try this:
https://send.firefox.com/

jpburnett, 2018-01-16:
https://send.firefox.com/download/daf3ca4e32/#qNoGon7o_Avje-ci3ld17g

vpetersson, 2018-01-17:
@antonmolodykh pls take a look at this one.

vpetersson, 2018-01-18:
I think I've finally run into this myself.

antonmolodykh, 2018-01-23:
@jpburnett Do I understand correctly that you have four devices, with three of them having a problem, and the fourth one is all right? Is it still like this?

jpburnett, 2018-01-23:
At this point all have experienced the issue. I was unaware of the keyboard being able to wake up the screen, so I am unsure if it was due to the viewer crashing and unable to start or if it is because the screen went to sleep.

antonmolodykh, 2018-01-24:
ok, the main difficulty for me is that I still don't understand how to reproduce the problem. I have seen this several times under different circumstances, but I still don't understand how I can cause this problem intentionally, because of this, it is more difficult to solve the problem.
@jpburnett you sent us an archive with your assets. can you describe in detail when the problem arises? after what time does the screen darken? are you still doing something with the device at this time (for example using ssh)?

jpburnett, 2018-01-26:
The screens darken after different times really, which is why its frustrating for me to know how to describe it. Some of the screens stay on longer than others. If I re-image the pi sometimes they last a day or two, maybe more. But every time I do this they end up going dark. I enable SSH but I never use it besides to reboot the pi if the screen darkens.
The assets we use are only webpages from a tableau server of data graphs, so i don't think its anything too demanding. I haven't tried using any other assets really with exception to the provided/preinstalled ones on the image (the weather and clock).

antonmolodykh, 2018-01-29:
@jpburnett ok, i'm still dealing with the problem. Can you do `sudo journalctl > foobar.txt`? Maybe this will help me. You already sent logs, but it seems then there was some other problem not related to this, so those logs don't carry any useful information.

jpburnett, 2018-01-30:
[screenReport.txt](https://github.com/Screenly/screenly-ose/files/1679409/screenReport.txt)
@antonmolodykh Here is the journalctl from one of the systems. Let me know if there is anything else I need to try.

antonmolodykh, 2018-02-01:
@jpburnett I made a fix, I think this should fix the problem. Please run the update (development version) and let me know if there will be problems. Thanks for your help.