Screenly/Anthias #708: web management console not working

bdorminy, 2017-12-12:
I saw that there was an issue a while back regarding this, but after upgrading to the latest version using, bash <(curl -sL https://www.screenly.io/install-ose.sh), I can no longer browse to the management console. The upgrade appeared to work fine and then after the reboot, the console was inaccessible. This has happened on 2 different devices. The display still works with the assets, I can ping the devices, and SSH into them, but no web admin. Any advice?

vpetersson, 2017-12-14:
If you upgraded to the latest development version, the port did change from :8080 to :80 (unless you're using SSL). Hence try just going to `http://[ip of your device]`.

bdorminy, 2017-12-14:
I did upgrade to the latest dev version, but it won't let me browse to port 80 either.

antonmolodykh, 2017-12-14:
please try again with another browser (or for example the incognito mode if you use chrome). if you once used `https://ip`, then the browser could remember this and now try to open `https` instead of `http`

bdorminy, 2017-12-14:
Thanks Anton. I have tried multiple browsers and chrome incognito with same results. No connection. What file is this setting in? Maybe I can look at the configuration file and see something there.

antonmolodykh, 2017-12-14:
Hm strange. Could you show me the output of the commands `cat /home/pi/.screenly/screenly.conf` and `cat /etc/nginx/sites-enabled/screenly.conf`?

bdorminy, 2017-12-14:
$ cat /home/pi/.screenly/screenly.conf
[viewer]
verify_ssl = off
default_streaming_duration = 300
player_name = SLAC
debug_logging = off
show_splash = off
resolution = 1920x1080
audio_output = hdmi
shuffle_playlist = off
use_24_hour_clock = off
default_duration = 10

[main]
use_24_hour_clock = off
websocket_port = 9999
listen = 0.0.0.0:8080
assetdir = /home/pi/screenly_assets
database = /home/pi/.screenly/screenly.db

[auth]
password =
user =

$ cat /etc/nginx/sites-enabled/screenly.conf
cat: /etc/nginx/sites-enabled/screenly.conf: No such file or directory

antonmolodykh, 2017-12-14:
@bdorminy now it seems even more strange to me than before. 
Could you show me the output of `cat /home/pi/screenly/settings.py`?

bdorminy, 2017-12-14:
Thanks @antonmolodykh. Here is the output.

$ cat /home/pi/screenly/settings.py
#!/usr/bin/env python
# -*- coding: utf8 -*-

from os import path, getenv
from sys import exit
from time import sleep
import ConfigParser
import logging
from UserDict import IterableUserDict
from flask import request, Response
from functools import wraps
import zmq

CONFIG_DIR = '.screenly/'
CONFIG_FILE = 'screenly.conf'
DEFAULTS = {
    'main': {
        'database': CONFIG_DIR + 'screenly.db',
        'listen': '0.0.0.0:8080',
        'assetdir': 'screenly_assets',
        'use_24_hour_clock': False,
        'websocket_port': '9999'
    },
    'viewer': {
        'player_name': '',
        'show_splash': True,
        'audio_output': 'hdmi',
        'shuffle_playlist': False,
        'resolution': '1920x1080',
        'default_duration': '10',
        'default_streaming_duration': '300',
        'debug_logging': False,
        'verify_ssl': True,
    },
    'auth': {
        'user': '',
        'password': ''
    }
}
CONFIGURABLE_SETTINGS = DEFAULTS['viewer']
CONFIGURABLE_SETTINGS['use_24_hour_clock'] = DEFAULTS['main']['use_24_hour_clock']

# Initiate logging
logging.basicConfig(level=logging.INFO,
                    format='%(message)s',
                    datefmt='%a, %d %b %Y %H:%M:%S')

# Silence urllib info messages ('Starting new HTTP connection')
# that are triggered by the remote url availability check in view_web
requests_log = logging.getLogger("requests")
requests_log.setLevel(logging.WARNING)

logging.debug('Starting viewer.py')

class ScreenlySettings(IterableUserDict):
    """Screenly OSE's Settings."""

    def __init__(self, *args, **kwargs):
        IterableUserDict.__init__(self, *args, **kwargs)
        self.home = getenv('HOME')
        self.conf_file = self.get_configfile()

        if not path.isfile(self.conf_file):
            logging.error('Config-file %s missing', self.conf_file)
            exit(1)
        else:
            self.load()

    def _get(self, config, section, field, default):
        try:
            if isinstance(default, bool):
                self[field] = config.getboolean(section, field)
            elif isinstance(default, int):
                self[field] = config.getint(section, field)
            else:
                self[field] = config.get(section, field)
        except ConfigParser.Error as e:
            logging.debug("Could not parse setting '%s.%s': %s. Using default value: '%s'." % (section, field, unicode(e), default))
            self[field] = default
        if field in ['database', 'assetdir']:
            self[field] = str(path.join(self.home, self[field]))

    def _set(self, config, section, field, default):
        if isinstance(default, bool):
            config.set(section, field, self.get(field, default) and 'on' or 'off')
        else:
            config.set(section, field, unicode(self.get(field, default)))

    def load(self):
        """Loads the latest settings from screenly.conf into memory."""
        logging.debug('Reading config-file...')
        config = ConfigParser.ConfigParser()
        config.read(self.conf_file)

        for section, defaults in DEFAULTS.items():
            for field, default in defaults.items():
                self._get(config, section, field, default)
        try:
            self.get_listen_ip()
            int(self.get_listen_port())
        except ValueError as e:
            logging.info("Could not parse setting 'listen': %s. Using default value: '%s'." % (unicode(e), DEFAULTS['main']['listen']))
            self['listen'] = DEFAULTS['main']['listen']

    def save(self):
        # Write new settings to disk.
        config = ConfigParser.ConfigParser()
        for section, defaults in DEFAULTS.items():
            config.add_section(section)
            for field, default in defaults.items():
                self._set(config, section, field, default)
        with open(self.conf_file, "w") as f:
            config.write(f)
        self.load()

    def get_configdir(self):
        return path.join(self.home, CONFIG_DIR)

    def get_configfile(self):
        return path.join(self.home, CONFIG_DIR, CONFIG_FILE)

    def get_listen_ip(self):
        return self['listen'].split(':')[0]

    def get_listen_port(self):
        return self['listen'].split(':')[1]

    def check_user(self, user, password):
        if not self['user'] or not self['password']:
            logging.debug('Username or password not configured: skip authentication')
            return True

        return self['user'] == user and self['password'] == password

settings = ScreenlySettings()

class ZmqPublisher:
    INSTANCE = None

    def __init__(self):
        if self.INSTANCE is not None:
            raise ValueError("An instantiation already exists!")

        self.context = zmq.Context()
        self.socket = self.context.socket(zmq.PUB)
        self.socket.connect('tcp://127.0.0.1:10001')
        sleep(1)

    @classmethod
    def get_instance(cls):
        if cls.INSTANCE is None:
            cls.INSTANCE = ZmqPublisher()
        return cls.INSTANCE

    def send(self, msg):
        self.socket.send(msg)

def authenticate():
    return Response("Access denied", 401, {"WWW-Authenticate": "Basic realm=private"})

def auth_basic(orig):
    @wraps(orig)
    def decorated(*args, **kwargs):
        if not settings['user'] or not settings['password']:
            return orig(*args, **kwargs)
        auth = request.authorization
        if not auth or not settings.check_user(auth.username, auth.password):
            return authenticate()
        return orig(*args, **kwargs)
    return decorated

antonmolodykh, 2017-12-14:
@bdorminy 
now I can confidently say that the update did not happen or something went wrong. were there any errors during the update?
in any case, I recommend that you run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` again.
please let me know here about the results

bdorminy, 2017-12-14:
@antonmolodykh 
Here is the results. There does appear to be some failure toward the end.

$ bash <(curl -sL https://www.screenly.io/install-ose.sh)
   _____                           __         ____  _____ ______
  / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
  \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
 ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
/____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                 /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Would you like to use the development branch? You will get the latest features, but things may break. (y/N)

Would you like to perform a full system upgrade as well? (y/N)
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done
+ sudo apt-get purge -y python-setuptools python-pip python-pyasn1
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'python-pyasn1' is not installed, so not removed
Package 'python-setuptools' is not installed, so not removed
Package 'python-pip' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 25 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up install-info (6.3.0.dfsg.1-1+b1) ...
Not a directory: /usr/share/info.
dpkg: error processing package install-info (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 install-info
E: Sub-process /usr/bin/dpkg returned an error code (1)
+ sudo apt-get install -y python-dev git-core libffi-dev libssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
git-core is already the newest version (1:2.11.0-3+deb9u2).
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0f-3+deb9u1).
python-dev is already the newest version (2.7.13-2).
0 upgraded, 0 newly installed, 0 to remove and 25 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up install-info (6.3.0.dfsg.1-1+b1) ...
Not a directory: /usr/share/info.
dpkg: error processing package install-info (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 install-info
E: Sub-process /usr/bin/dpkg returned an error code (1)
+ curl -s https://bootstrap.pypa.io/get-pip.py
+ sudo python
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages
+ sudo pip install ansible==2.1.0.0
Requirement already satisfied: ansible==2.1.0.0 in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: PyYAML in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0)
Requirement already satisfied: setuptools in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0)
Requirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0)
Requirement already satisfied: paramiko in /usr/local/lib/python2.7/dist-packages (from ansible==2.1.0.0)
[... 85 lines cut ...]

TASK [system : Remove build-essential] *****************************************
fatal: [localhost]: FAILED! => {"changed": false, "failed": true, "msg": "'apt-get remove 'build-essential'' failed: E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr": "E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following packages were automatically installed and are no longer required:\n  g++ g++-6 libstdc++-6-dev\nUse 'sudo apt autoremove' to remove them.\nThe following packages will be REMOVED:\n  build-essential\n0 upgraded, 0 newly installed, 1 to remove and 25 not upgraded.\n1 not fully installed or removed.\nAfter this operation, 20.5 kB disk space will be freed.\nSetting up install-info (6.3.0.dfsg.1-1+b1) ...\r\nNot a directory: /usr/share/info.\r\ndpkg: error processing package install-info (--configure):\r\n subprocess installed post-installation script returned error exit status 1\r\nErrors were encountered while processing:\r\n install-info\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following packages were automatically installed and are no longer required:", "  g++ g++-6 libstdc++-6-dev", "Use 'sudo apt autoremove' to remove them.", "The following packages will be REMOVED:", "  build-essential", "0 upgraded, 0 newly installed, 1 to remove and 25 not upgraded.", "1 not fully installed or removed.", "After this operation, 20.5 kB disk space will be freed.", "Setting up install-info (6.3.0.dfsg.1-1+b1) ...", "Not a directory: /usr/share/info.", "dpkg: error processing package install-info (--configure):", " subprocess installed post-installation script returned error exit status 1", "Errors were encountere [... cut ...]
+ sudo rm -rf /usr/share/man /usr/share/groff /usr/share/info /usr/share/lintian /usr/share/linda /var/cache/man
+ sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -exec rm -r '{}' ';'
+ cd /home/pi/screenly
+ git rev-parse HEAD
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)

bdorminy, 2017-12-14:
@antonmolodykh after doing a new install and NOT choosing the dev option, the web administration is working.

antonmolodykh, 2017-12-14:
thanks, @bdorminy 
how is the Screenly installed? Did you use our image or installation through NOOBS?

bdorminy, 2017-12-14:
Originally through NOOBS.

antonmolodykh, 2017-12-15:
@bdorminy 
unfortunately I could not reproduce it. I installed Screenly through NOOBS and launched the update, everything works fine.
looking at your setup logs it seems that something is wrong in the system. I recommend that you download the latest NOOBS, install Screenly and try to run the update again

vpetersson, 2017-12-15:
Judging by the logs, I wouldn't rule out that it's a broken SD card. I would do one of the two following as a next step:

 * Replace the SD card
 * Download [SD Formatter](https://www.sdcard.org/downloads/formatter_4/) and do an overwrite format

bdorminy, 2017-12-15:
Thanks @vpetersson and @antonmolodykh. It's odd that it happened on 2 identical devices. I'm going to do some testing and try installing with different options, as well as getting a fresh copy of NOOBS. Thanks again for the feedback and if I come across anything significant I'll post it.

realparker, 2017-12-29:
Just did a fresh install based on a raspberry 2 running stretch lite. Installation with curl took ages because of pip (nearly 5 hours) but I've been patient enough to let it end with success.
When I then first rebooted the pi, the attached TV only shows the screenly page "no active network connection found" altough the pi is pingable. I saw a post in a forum with a similar issue but unfortunately without an answer for more than a month.
I also can't connect to the management console.
Connection is via WLAN USB dongle and properly working (pingable & ssh-able).

```
$ cat /etc/network/interfaces
# Generated by Screenly Network Manager
auto lo
  iface lo inet loopback

auto eth0
  iface eth0 inet dhcp
```

```
$ ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:df:07:b8  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 66  bytes 140122 (136.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 66  bytes 140122 (136.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.178.66  netmask 255.255.255.0  broadcast 192.168.178.255
        inet6 fe80::bd0:fa5a:da48:15a  prefixlen 64  scopeid 0x20<link>
        ether 74:da:38:33:84:1f  txqueuelen 1000  (Ethernet)
        RX packets 1291  bytes 234828 (229.3 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 643  bytes 98924 (96.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

When I connect through eth0 everything is working properly as supposed and http://#myip#:8080 is fine. What did I miss (I'm a bit afraid this could be a well known issue debian-strech based...)?

Cheers, Richie

vpetersson, 2017-12-29:
@realparker 

> Installation with curl took ages because of pip (nearly 5 hours) but I've been patient enough to let it end with success.

Must have been a very slow SD card (perhaps Class 4) or a very slow connection. It's slow (normally ~30min - 1h), but not that slow. :)

> What did I miss (I'm a bit afraid this could be a well known issue debian-strech based...)?

Kind of. Stretch did introduce a fair amount of changes, and so did we on our end. I presume you installed the production version. We're now on the final stretch of the [next release](https://github.com/Screenly/screenly-ose/milestone/8) which will (hopefully) improve the networking a bit further.

I'd suggest that you try upgrading to the development release and see if that solves the issue for you. That will also change a bit on how networking is managed (which we will cover in more details in the release blog post).

xlegs, 2018-01-17:
I am having the same issue with dev on a fresh raspbian lite install. Networking is OK as nginx and couchdb are functioning properly. 

/home/pi/screenly/settings.py:
```
#!/usr/bin/env python
# -*- coding: utf8 -*-

from os import path, getenv
from sys import exit
from time import sleep
import ConfigParser
import logging
from UserDict import IterableUserDict
from flask import request, Response
from functools import wraps
import zmq

CONFIG_DIR = '.screenly/'
CONFIG_FILE = 'screenly.conf'
DEFAULTS = {
    'main': {
        'database': CONFIG_DIR + 'screenly.db',
        'assetdir': 'screenly_assets',
        'use_24_hour_clock': False,
        'websocket_port': '9999',
        'use_ssl': False,
        'analytics_opt_out': False
    },
    'viewer': {
        'player_name': '',
        'show_splash': True,
        'audio_output': 'hdmi',
        'shuffle_playlist': False,
        'resolution': '1920x1080',
        'default_duration': '10',
        'default_streaming_duration': '300',
        'debug_logging': False,
        'verify_ssl': True,
    },
    'auth': {
        'user': '',
        'password': ''
    }
}
CONFIGURABLE_SETTINGS = DEFAULTS['viewer']
CONFIGURABLE_SETTINGS['use_24_hour_clock'] = DEFAULTS['main']['use_24_hour_clock']

PORT = int(getenv('PORT', 8080))
LISTEN = getenv('LISTEN', '127.0.0.1')

# Initiate logging
logging.basicConfig(level=logging.INFO,
                    format='%(message)s',
                    datefmt='%a, %d %b %Y %H:%M:%S')

# Silence urllib info messages ('Starting new HTTP connection')
# that are triggered by the remote url availability check in view_web
requests_log = logging.getLogger("requests")
requests_log.setLevel(logging.WARNING)

logging.debug('Starting viewer.py')

class ScreenlySettings(IterableUserDict):
    """Screenly OSE's Settings."""

    def __init__(self, *args, **kwargs):
        IterableUserDict.__init__(self, *args, **kwargs)
        self.home = getenv('HOME')
        self.conf_file = self.get_configfile()

[... 123 lines cut ...]
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:5984            0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:5986          0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      444/python          
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:4369            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:42713           0.0.0.0:*               LISTEN      -                   
tcp6       0      0 :::9999                 :::*                    LISTEN      211/python          
tcp6       0      0 :::80                   :::*                    LISTEN      -                   
tcp6       0      0 :::4369                 :::*                    LISTEN      -                   
tcp6       0      0 :::21                   :::*                    LISTEN      -                   
tcp6       0      0 :::22                   :::*                    LISTEN      -                   
udp        0      0 0.0.0.0:58088           0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:68              0.0.0.0:*                           -                   
udp6       0      0 :::53978                :::*                                -                   
udp6       0      0 :::5353                 :::*                                -    
```
I moved /etc/nginx/sites-enabled/screenly.conf to sites-available because it conflicted with my nginx setup.

Any ideas?

vpetersson, 2018-01-18:
@xlegs Looks like the service is running just fine, but if you're intending to run more than Screenly OSE through Nginx, you need to do some hacking in the config files as it will not work out-of-the-box because with two different setups on the same port. 

You need to do one of the following things:

 * Create a vhost such as screenly.mydomain.local and point that DNS record to your device, and then subsequently patch the Nginx file to listen for that vhost.
 * Patch Nginx to listen on an alternative port, such as 0.0.0.0:81 and use that for OSE (or for the other service)
 * Patch Nginx to route something like `/screenly` to the port. 

In summary, this is not a Screenly issue per se, but rather a configuration in Nginx.

closed 2018-10-19 by stale[bot]